dartsim/dart #317: Conflict of joint friction and joint position limit

jslee02, 2015-01-26:
DART cannot handle joint constraints correctly when there exists joint friction and joint position limit at the same time. This is because the two constraints overwrite the constraint impulse rather than adding up.

closed 2016-01-07 by jslee02

scpeters, 2016-01-22:
Thanks for fixing this and back-porting to 4.3. I've tested with gazebo and seen an improvement, though I had to relax the tolerances slightly (1e-4 to 1e-3).

* https://bitbucket.org/osrf/gazebo/commits/60614007ffa909f01399c3478707a85b7069bb61

Do you have any ideas why to tolerances were slightly violated? Constraint error correction?

jslee02, 2016-01-22:
The main reason is that DART doesn't correct positional violation directly. Instead, DART uses artificial velocity with ERP (error reduction parameter) so that the position error is corrected by integrating the position with the velocity. If the ERP is 1.0 then position will be completely corrected, and if ERP is 0 then position will not be corrected. The default ERP is 0.01. If we increase this ERP then it might reduce the tolerance.

I picked the low value for the default ERP, which may not correct the error aggressively, because too much ERP would make the system instable.

To handle this more reliably, DART may need to solve LCP twice per simulation for velocity correction and position correction per every simulation step like ODE does unless we change the whole constraint handling framework to other than impulse-based method.

commit c6538357 (2016-01-07): Fix incorrect applying constraints on a joint with different constraints
- Each joint constraint impulse is added up to the joint instead of overwriting the total impulse everytime.
- This should resolve #317 as well.
PR #579 (2016-01-08): Fix incorrect applying of joint constraint impulses --  Patch for 5.1
This pull request fixes #317 for DART 5.1 as fixed in #566 for DART 6.0.
PR #578 (2016-01-08): Fix incorrect applying of joint constraint impulses --  Patch for 4.3
This pull request fixes #317 for DART 4.3 as fixed in #566 for DART 6.0.