dartsim/dart #330: Bug in World->removeSkeleton()

ana-GT, 2015-02-11:
Hi, here a report on the following bug:

Function: removeSkeleton()
Location: dart/simulation/World.cpp (line 246)
Error: The function should delete the skeleton name from the mNameMgrForSkeletons, which is not currently being done.
Fix: Add the following 2 lines probably after line 265 for safe checks:

// Remove from name manager
  mNameMgrForSkeletons.removeName( _skeleton->getName() );

I am using a program in which I constantly add/delete skeletons with the same name for some online object detection. I was getting SEGFAULTS after deleting an skeleton and then trying to add it later (the NameManager thought the object still existed, but the Skeleton had been already erased). 

After adding the couple of lines mentioned above, the issue dissapared.

mxgrey, 2015-02-11:
Good catch. I'll throw in a fix immediately, and make a regression test for
this.

mxgrey, 2015-02-11:
Pull request #331 now addresses this issue.

closed 2015-02-14 by mxgrey