dartsim/dart #333: Bug in FreeJoint helper

ana-GT, 2015-02-13:
* File: dart/dynamics/FreeJoint.cpp
* Functions compromised: convertToPositions (line 61) and converToTransform (line 70)
* Problem description: I am generating a set of transformations with rotations around Z with angles in the range  [0,2*PI], increasing at each step. I use these Transformations to set the joint positions of a freeJoint (using the same formula as used in convertToPositions). 

The joint positions I get are wrong for some angles. The joint positions correspond to the intended Transformation rotated 90 around Z. This happens usually after half of the range (either if I use [0,2PI] or [-PI,PI]. An example in which the bug happens is copied at the end of this issue. In this example I use:

Input transform:

-0.815214  -0.57916         0  0.437796
 -0.57916  0.815214         0  0.488175
        0         0        -1      1.03
        0         0         0         1

Positions obtained using convertToPositions():
-0.954925  -2.99295        -0  0.437796  0.488175      1.03

Output transform obtained using convertToTransform() (should be the same as input transform):
   -0.815214      0.57916 -5.39747e-16     0.437796
     0.57916     0.815214  1.72211e-16     0.488175
 5.39747e-16 -1.72211e-16           -1         1.03
6.95317e-310 1.38338e-322 6.92085e-310 2.12999e-317

Notice the flipped signs in x an y columns.

The problem described above is related to convertToPositions. I also noticed that in convertToTransform, there is a line missing: Tf.setIdentity(). Otherwise, the last row of Tf will be filled with garbage (as in the example above).

* Final note: Please, if somebody can fix this quickly I would appreciate it. I spent most of today searching for the bug in  my own code until I realized that the bug was (very possibly and unless I am mistaken) in dart so I would like to continue with my work once the bug is fixed. Thanks!

PS.- Bug test file:

#include <dart/math/Geometry.h>
#include <iostream>

Eigen::Vector6d convertToPositions(const Eigen::Isometry3d& _tf);
Eigen::Isometry3d convertToTransform(const Eigen::Vector6d& _positions);

int main( int argc, char* argv[] ) {

  Eigen::Isometry3d Tf;
  Eigen::VectorXd q;
  Eigen::Isometry3d Tf_back;

  Tf.setIdentity();
  Tf.translation() << 0.1, 0.2, 0.3;

  double dang = 10*M_PI/180.0;
  int n = (int) (2*M_PI /dang);

  Tf.matrix() << -0.815214,  -0.57916,        -0,  0.437796,
 	-0.57916,  0.815214,         0,  0.488175,
        0,         0,        -1,      1.03,
        0,         0,         0,         1;

   double det = Tf.linear().determinant();
  std::cout << "Determinant of test TF: "<< det << std::endl;

/*
  for( size_t i = 0; i < n; ++i ) {
    Eigen::Matrix3d rot;
    rot = Eigen::AngleAxisd( i*dang, Eigen::Vector3d(0,0,1) );
    Tf.linear() = rot;
*/
    // Get position with helper function
    q = convertToPositions( Tf );  

    // Get translation back with helper function
    Tf_back = convertToTransform( q );
    //std::cout << "i: "<< i << std::endl;
    std::cout << "\n ** Tf original: \n"<< Tf.matrix() << std::endl;
    std::cout << "\n ** Positions from Tf: \n"<< q.transpose() << std::endl;
    std::cout << "\n ** Tf from q: \n" << Tf_back.matrix() << std::endl;
    std::cout << "\n\n" << std::endl;

/*
  }
*/

}

//==============================================================================
Eigen::Vector6d convertToPositions(const Eigen::Isometry3d& _tf)
{
  Eigen::Vector6d x;
  x.head<3>() = dart::math::logMap(_tf.linear());
  x.tail<3>() = _tf.translation();
  return x;
}

//==============================================================================
Eigen::Isometry3d convertToTransform(
    const Eigen::Vector6d& _positions)
{
  Eigen::Isometry3d tf;
  tf.linear() = dart::math::expMapRot(_positions.head<3>());
  tf.translation() = _positions.tail<3>();
  return tf;
}

mxgrey, 2015-02-14:
Indeed, it looks like the full cycle was never put in a unit test. I'm working to fix this right now.

mxgrey, 2015-02-14:
I was under the mistaken impression that Eigen::Isometry3d behaves like Eigen::Affine where the bottom row is assumed to be 0 0 0 1 and therefore does not need to be set. Looking at Eigen/Geometry/Transform.h, apparently that's not the case (although I have no idea why).

Is adding tf.setIdentity() sufficient to fix the issues you're observing? I've added a test that cycles through convertToPostions -> convertToTransform and checks that the new transform matches the original transform. The test indeed failed with the original code, but it succeeds now by passing Eigen::Isometry3d::Identity() into the constructor of **tf** in the function convertToTransform(). I can't think of any other fixes that would be needed.

mxgrey, 2015-02-14:
If there are no other issues, pull request #334 now covers this issue.

mxgrey, 2015-02-14:
I've made a version of your test code, and I see that this sign flipping issue still exists. I'll continue to investigate that.

ana-GT, 2015-02-14:
Grey,

Great. Thanks a lot.

Just to give you an idea how I am running into this error: I am generating
overhead grasps for a cylinder. The grasp is performed by a hand skeleton.
The grasp is defined with a Transformation such that the Z axis of the hand
points "Down" (0,0,-1) and the X axis (simulating the finger closing
direction) rotates from 0 to 2PI representing how the hand rotates around Z.

When I visualize the grasps I notice that:

1. From 0 to 90:  The hand moves as expected, it rotates around Z one step
at a time
2. From 90 to ~180: The hand moves backwards (from 90 to 0) instead of keep
moving from 90 to 180
3. From 180 to 360 moves back and forth (I don't remember exactly, I think
from 0 to -90 and then comes back to 0).

Maybe a good test would be to generate a set of rotations around Z from 0
to 2PI and compare the input rotation part of the transformation with the
output rotation after using the toPosition / toTransformation functions.
Probably using a quaternion to compare the rotations would be easier.

Thanks again for taking care of this, I appreciate it.

Best

Ana

mxgrey, 2015-02-14:
Alright, I believe I may have found a bug in the pi-singularity handling of logMap.

It seems that it does not always produce the correct signs for the components in the vector it returns. As a result, a pi-rotation around an axis can give back a position vector that represents a pi-rotation around a reflection of the original axis.

I'm going to investigate what can be done to correct this, but I don't know what the original reference for the code in DART is, so I don't know how long it might take to come up with a correction.

mxgrey, 2015-02-14:
@ana-GT I have a working solution for the problem you've encountered, but I'd like to put together a formal regression test for it.

In my attempts to put together a regression test, I have not been able to reproduce the error that you experienced by using formally generated rotation matrices (in other words, using tools like Eigen::AngleAxis to create a well-formed rotation matrix). This makes me suspect that the issue only occurs if a hard-coded (and therefore poorly conditioned) rotation matrix is used somewhere in the mix. Have you encountered a case where this issue occurs when you're not using a hard-coded matrix? If so, I'd like to write a test around it.

If not, the solution that I've come up with is going to be more robust to hard-coded and imprecise matrices, while also being faster. Of course the real credit for it goes to Eigen, since I'm just exploiting some built-in Eigen features.

mxgrey, 2015-03-19:
I believe this issue has been resolved.

closed 2015-03-19 by mxgrey