dartsim/dart #336: Dart not compiling

marijnfs, 2015-02-20:
Dart doesn't compile on my machine (Mint, based on Ubuntu Trusty, with latest git head: c546045389a46bbe366ffc3ce2c0f8b571e7facf ). It  gives the following error:

from /home/marijnfs/software/dart/unittests/testCollision.cpp:50:
/home/marijnfs/software/dart/dart/utils/urdf/urdf_world_parser.h:45:3: error: ‘World’ does not name a type
   World* parseWorldURDF(const std::string &xml_string, std::string _path );

mxgrey, 2015-02-20:
Do you know if you have URDF headers installed?

marijnfs, 2015-02-21:
Yes they seem to be installed.

mxgrey, 2015-02-24:
Is this the only compilation error that you're seeing so far? I've run out of guesses for what could be causing it.

Are you familiar with header guards? It looks like urdf_world_parser.h uses \#pragma once as a header guard, which I don't believe works on all platforms (I'm pretty sure that's only meant for Windows). You can try changing the file to use a more standard header guard:

```
#ifndef DART_UTILS_URDF_URDF_WORLD_PARSER
#define DART_UTILS_URDF_URDF_WORLD_PARSER

// content of the header

#endif // DART_UTILS_URDF_URDF_WORLD_PARSER
```

And let us know if that fixes the compilation issue. I don't really expect it to, because I would expect a different error message if it were a header guard issue, but I don't have any other ideas about what might be strange/wrong with that header.

mxgrey, 2015-07-03:
@marijnfs Are you still having this issue? I haven't heard about anyone else running into this problem, so I'm inclined to close this unless it's still occurring.

marijnfs, 2015-07-03:
Let me try tomorrow

marijnfs, 2015-07-06:
I don't have time now to go through installation, i think you can close 
it for now.

jslee02, 2016-01-31:
Closing this issue. Please reopen if the error still exists.

closed 2016-01-31 by jslee02