dartsim/dart #337: Use special prefix for autoupdating member variables

jslee02, 2015-02-21:
This issue came up from #319. Let's continue the discussion here.

---

The original issue description:

@mxgrey 

It's important to note that the auto-updating comes with one important caveat for all DART developers now and in the future: there are certain data members that must not be modified or even accessed outside of the appropriate functions, even for internal use. Some of these include BodyNode::mWorldTransform, BodyNode::mVelocity, BodyNode::mAcceleration, and even others that might not be expected like Joint::mJacobian, Joint::mJacobianDeriv, BodyNode::mArtInertia. There are too many to list right here, but they should all be noted in the header comments. The reason these members should not be used directly is because they are carefully monitored by the auto-updating system, so anything that writes to them when it shouldn't will invalidate results, and anything that reads from them when it shouldn't might end up using outdated data.

This raises a question that I'd like to pose: Should we append something to the names of these "untouchable" data members similar to how the [Hungarian Notation](https://en.wikipedia.org/wiki/Hungarian_notation) encodes information in variable names? Normally I would not advocate such a naming convention, but it might help future (and even current) developers if there were a clear way of immediately knowing that they should not use a variable directly. I don't really have any good suggestions for what the code should be, though. Maybe append "_dnu" (for "do not use") to the ends of the variable names?

---

Additional comments:

@mxgrey 

What do people think of altering the names of any internal variables that must not be used outside of their set/get functions? The auto-updating features result in really good robustness to user error, but this is at the cost of greater sensitivity to developer error, because if a developer haphazardly uses an auto-updating variable outside of its set/get/update functions, then it will almost certainly produce wrong results.

To prevent that, I would propose adopting some naming convention that makes it clear which variables are safe to use internally and which aren't. I think the best thing I've come up with for the moment would be a prefix of "_au" where "au" stands for "auto-update" and the underscore makes sure that the variable names only show up at the very end of an autocomplete list (assuming your autocomplete sorts variable names alphabetically). For example, since mWorldTransform is an auto-updated variable, we would change its name to _auWorldTransform. This way, it can be immediately clear to all current and future developers whether a variable is safe to use internally (because it would start with an "m") or whether it can only be interacted with through set/get functions (because it would start with an "_au").

What does everyone think?

@jslee02

Regarding the special naming for auto-updating variables, I have have questions. All the auto-updating variables are modified by only the declared classes not by inheriting classes? If so, how about making them as private member variables? If it's possible, then it would be safer way.

@mxgrey 

> All the auto-updating variables are modified by only the declared classes not by inheriting classes?

Not always. For example, Joint::mLocalTransform is an auto-updated member, but it needs to be set by the updateTransform() function which is virtual and gets overridden by each joint type. So Joint::mLocalTransform and almost all of the other Joint members cannot be made private.

Also, privatizing them doesn't make them safe from misuse within the class that they belong to, which is my biggest concern. I think it's natural for a developer to assume that they have the freedom to directly use any member variables of the class that they're working on. Also, there are many member variables that *can* be used directly with no negative consequences, because they don't auto-update. So I think a naming convention would help to differentiate which variables can and cannot be used directly. That should help prevent a lot of confusion and second-guessing while developers are working on a given class.

@jslee02

If auto-updating variables should be interacted through get/set functions, then it is necessary to make sure which functions are allowed to modify the variables. If not, it is still prone to be misused. It seem like get function is the allowed function, right? For example, auWorldTransform is must be modified by only getWorldTransform(). 

Also, underscore is preserved for parameter name so I would like to use "au" for the auto-updating variables.

@mxgrey 

> It seem like get function is the allowed function, right?

In some cases, but not always. For example, mArticulatedInertia is now an auto-updated variable, but it can't be updated in a single "get" function, because it's updated in a Skeleton class function that iterates through all the BodyNodes. Within that Skeleton function, the BodyNode::mArticulatedInertia members need to be directly accessed and changed. So we do need to be aware of which specific functions can alter the values of which specific data members. I already have that worked out at the moment, but maybe I should put it into documentation somewhere for posterity?

> Also, underscore is preserved for parameter name so I would like to use "au" for the auto-updating variables.

Yeah, this is something that I kind of wanted to talk about (although I was going to save it for later). I've never quite understood why parameter names are given underscores. I would understand it if member variables didn't have any designator, because an underscore in parameter names would ensure that there is no conflict between parameter names and member variable names, but all member variables start with an "m", so that should already prevent any conflicts. In most conventions that I've seen, a leading underscore is used to indicate a protected/private member variable, so I find it awkward that they're being used for parameter names. I also feel like it hurts auto-complete readability a little bit (when seeing a preview of the function declaration in an IDE), but that might just be a personal issue.

In any case, the main reason that I'd rather not start them with just "au" is that I think most auto-completers will list suggestions in alphabetical order, so things that start with "au" would be at the very top of the list, but we're talking about variables that we want to make sure people *don't* use in general, so it would seem better to me if they were hidden away at the bottom. Underscore is usually sorted to the bottom of an alphabetical list, so leading with an underscore would ensure that those variables are relatively hidden.

psigen, 2015-06-13:
Would it make sense to put auto-updating member variables into a private member pure-data struct, e.g. `this.mAutoUpdated.mVelocity`?  (Fix naming convention as appropriate)

That would keep them scoped and less visible in autocomplete, semantically denote why they shouldn't be directly accessed in the general case, and not increase the complexity of accessing the data within the class.

Just a random thought.

mxgrey, 2015-06-13:
I definitely like the idea of putting them in a specially named struct. However, they have to be protected rather than private, because some of them (like articulated inertia for SoftBodyNodes) may need to be handled in a special way by an inheriting class.

closed 2025-11-03 by jslee02