dartsim/dart #340: Problem in running examples

KooroshNaderi, 2015-02-24:
Dear all,

hi, my name is Kourosh Naderi. I am new user of DART simulator. I have installed DART 4.1.0 on my computer which runs windows 7 as its operating system. I used the instructions on <https://github.com/dartsim/dart/wiki/Windows-Installation%20for%20DART%204.3> page to install DART on my computer. However, the only example that I can run it without any problems is "forwardSim", and when I run the other samples I get the following error:
"Warning! Invalid GJK solver"
Then for example in "Vehicle" sample, the car just passes through the floor and continues to go downward. Can anybody help me in this matter?

My another problem is that in "forwardSim" sample, I cannot change the rate of showing scene, and it seems like I do not have any controls on the simulator! Can anybody explain to me how the simulator works?

regards
Koursoh

mxgrey, 2015-02-24:
I don't have much experience running DART on Windows (I'm mostly an Ubuntu user), but I can try to help. I have a few questions to start out:

Did you install DART from binaries, or did you grab the source code and compile it yourself?

Do you have Visual Studio 12 (2013)?

If you compiled it yourself, were there any cmake or compilation errors or warnings that you noticed?

KooroshNaderi, 2015-02-25:
hi, thanks for the reply.

1st. I got the source code here and compile it myself.
2nd. yes, I have Visual Studio 12 (2013).
3rd. you are right. I got many warnings when I wanted to build the project using Cmake 3.1.0. In the following I have brought the warnings:

- "CMake Warning at CMakeLists.txt:86 (message):
  NLOPT not found"
- "CMake Warning at CMakeLists.txt:96 (message):
  IPOPT not found"
- "CMake Warning at CMakeLists.txt:113 (message):
  BulletCollision not found, for bullet collision detector option, please
  install libbullet-dev."
- "CMake Warning (dev) at CMakeLists.txt:119 (set):
  Policy CMP0053 is not set: Simplify variable reference and escape sequence
  evaluation.  Run "cmake --help-policy CMP0053" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
  For input:
    '@CMAKE_INSTALL_PREFIX@/include'
  the old evaluation rules produce:
    'C:/Golems/include'
  but the new evaluation rules produce:
    '@CMAKE_INSTALL_PREFIX@/include'
  Using the old result for compatibility since the policy is not set.
This warning is for project developers.  Use -Wno-dev to suppress it."

As you said it, the instructions are for Win32. So I have installed Boost 1.55 for Win32 and build DART 4.1.0 for Visual Studio 12 (Win32 ver). 
When I compiled the DART 4.1.0, I got lots of warnings too. Also, when I run each sample project of DART, I get a warning at first and when I press "space" to start the simulation, the simulation does not work properly and I get the following warning:
"Warning! Invalid GJK solver".

mxgrey, 2015-03-04:
Sorry for my delayed response. As I mentioned before, I never do development in Windows, so I've been struggling to teach myself how to set up Windows build environment properly. I had no trouble compiling the core-only version of DART, but the full version was giving me trouble. I was having boost issues for several days, only to realize that it was a combination of having the wrong version of boost installed and not having PATH set up correctly. Then I had some more PATH and linking issues... anyway, I'm able to build the full version of DART and all its apps on Windows now, which is nice.

When I compiled the 4.1 release branch, I experienced the exact same issue that you did: the car in the vehicle example fell through the floor.

I'm attempting to get the latest master to compile, but there seem to be undefined references to a gjk_libccd extension in FCL. I'm guessing the FCL binary that gets installed is out of date? I'll keep investigating.

But your issues are definitely being replicated, so this probably not an issue on your end.

jslee02, 2017-11-12:
Apologies. We don't maintain DART 4 anymore.

closed 2017-11-12 by jslee02