dartsim/dart #346: LineShape for easily creating line visualizations

mxgrey, 2015-03-12:
I'd like to create a LineShape that allows users to put lines into the Frame/Entity kinematic tree and be visualized. I don't anticipate making lines compatible with dynamic collision checking or handling, because they are 1D shapes, but I think it would be fair to use them as visualization shapes.

jslee02, 2015-03-12:
LineShape as visualization shape make sense but not for collision shape. First, it is not supported by collision checker. Although we can do collision check with LineShape, the dynamics can't be handled because of the zero momenta of inertia along the line's axis. For example, if we apply angular force along the axis of line then the acceleration would be infinite with the zero inertia. To support line dynamics model, new body node type would be required whose angular motion along the axis is ignored.

Anyways, LineShape as visualization shape sounds good. Please add error handling code for the case of attempting to add LineShape as collision shape.

mxgrey, 2015-03-12:
> Please add error handling code for the case of attempting to add LineShape as collision shape.

Should I just have it spit out a warning, or should I also refuse to allow LineShapes to be added as collision shapes?

jslee02, 2015-03-12:
Ignoring the attempting to add with a warning would be good.

jslee02, 2015-03-12:
I have a question. If it is able to collision check with LineShape then do you think it would be helpful for kinematics (or planning)? I have a plan to implement sensors for DART at some point. Then collision checking with ray should be implemented, and line can be supported easily.

mxgrey, 2015-03-12:
That's a really good point. Line segment collision checking could be used to simulate proximity sensors, or to take point samples of objects to simulate LIDAR. I could easily imagine that being very useful.

Then if we're supporting collision detection with line segments, we might want to once again consider whether it makes any sense to allow lines to be added to BodyNodes as collision shapes. I'm still strongly inclined to say no, because 1D lines are still fundamentally non-physical, and there might end up being edge cases that are difficult to handle.

mxgrey, 2015-03-12:
By the way, my plan was to implement line segments (I guess I should have been more clear about that). Do we also want to support infinite lines and rays as shapes?

jslee02, 2015-03-12:
If FCL can support line segment, it would be best to make line segment to work with other collision shapes. I'll take a look at it if I can do.

jslee02, 2015-03-12:
I think adding new shapes just for visualization is fine as long as we can draw it. So adding them sounds good to me.

mxgrey, 2015-03-12:
Off the top of my head, I'm not sure how to draw infinite rays or infinite lines in OpenGL or in OSG, but I can certainly look into it.

mxgrey, 2015-03-13:
I've implemented a LineSegmentShape, but I don't think the infinite ray/line is something that can be handled very gracefully by the visualizers. There's nothing built-in for OpenGL or OSG to support infinite geometries, so rendering of them would involve inspecting the view and drawing line segments from one extreme of the view to the other. It could be done, but I suspect it might be more effort than it's worth.

mxgrey, 2015-03-19:
Finished with #349

closed 2015-03-19 by mxgrey