dartsim/dart #351: missing bullet includes for DART 4.3

hsu, 2015-03-15:
Since DART depends on bullet collision, it should export bullet paths in its cmake config file:
~~~
diff --git a/cmake/DARTCoreConfig.cmake.in b/cmake/DARTCoreConfig.cmake.in
index 4e37dff..61778d3 100644
--- a/cmake/DARTCoreConfig.cmake.in
+++ b/cmake/DARTCoreConfig.cmake.in
@@ -11,6 +11,7 @@ get_filename_component(CURRENT_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
 set(DARTCore_INCLUDE_DIRS
   "@CMAKE_INSTALL_PREFIX@/include"
   "@Eigen_INCLUDE_DIRS@"
+  "@BULLET_INCLUDE_DIRS@"
 )
 
 set(DARTCore_LIBRARY_DIRS "@CMAKE_INSTALL_PREFIX@/lib" "@Boost_LIBRARY_DIRS@")
~~~

otherwise dependents might encounter errors such as below:
~~~
[ 43%] Building CXX object gazebo/physics/dart/CMakeFiles/gazebo_physics_dart.dir/DARTBallJoint.cc.o
cd /home/hsu/projects/haptix/gazebo_haptix/build_release/gazebo/physics/dart && /usr/bin/c++   -DLIBBULLET_VERSION=2.83 -Dgazebo_physics_dart_EXPORTS -s -O3 -DNDEBUG -msse4.2 -msse4.1 -mssse3 -msse3 -mfpmath=sse -msse -msse2  -std=c++11  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -fvisibility=hidden -fvisibility-inlines-hidden  -s -O3 -DNDEBUG -msse4.2 -msse4.1 -mssse3 -msse3 -mfpmath=sse -msse -msse2  -fPIC -I/home/hsu/projects/haptix/gazebo_haptix/test/gtest/include -I/home/hsu/projects/haptix/gazebo_haptix -I/home/hsu/projects/haptix/gazebo_haptix/build_release -I/usr/include/libusb-1.0 -isystem /usr/include/OGRE/RTShaderSystem -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /home/hsu/local/include/sdformat-3.0 -I/home/hsu/local_dart_4.3/include -I/usr/include/eigen3    -fPIC -o CMakeFiles/gazebo_physics_dart.dir/DARTBallJoint.cc.o -c /home/hsu/projects/haptix/gazebo_haptix/gazebo/physics/dart/DARTBallJoint.cc
In file included from /home/hsu/local_dart_4.3/include/dart/collision/bullet/bullet.h:3:0,
                 from /home/hsu/local_dart_4.3/include/dart/collision/collision.h:8,
                 from /home/hsu/projects/haptix/gazebo_haptix/gazebo/physics/dart/dart_inc.h:26,
                 from /home/hsu/projects/haptix/gazebo_haptix/gazebo/physics/dart/DARTJoint.hh:26,
                 from /home/hsu/projects/haptix/gazebo_haptix/gazebo/physics/dart/DARTBallJoint.hh:22,
                 from /home/hsu/projects/haptix/gazebo_haptix/gazebo/physics/dart/DARTBallJoint.cc:21:
/home/hsu/local_dart_4.3/include/dart/collision/bullet/BulletCollisionDetector.h:43:37: fatal error: btBulletCollisionCommon.h: No such file or directory
 #include <btBulletCollisionCommon.h>
                                     ^
compilation terminated.
make[2]: *** [gazebo/physics/dart/CMakeFiles/gazebo_physics_dart.dir/DARTBallJoint.cc.o] Error 1
make[2]: Leaving directory `/home/hsu/projects/haptix/gazebo_haptix/build_release'
make[1]: *** [gazebo/physics/dart/CMakeFiles/gazebo_physics_dart.dir/all] Error 2
make[1]: Leaving directory `/home/hsu/projects/haptix/gazebo_haptix/build_release'
make: *** [all] Error 2
~~~

jslee02, 2015-03-16:
Since bullet is an optional dependency of DART, I put the bullet includes under if statements that checks if bullet is installed.

jslee02, 2015-03-19:
I believe this is resolved by #352.

scpeters, 2015-03-21:
I think it's not actually fixed. The `HAVE_BULLET_COLLISION` variable isn't necessarily defined in the downstream code. The following will works for me (currently I can't build gazebo against dart without this change):
~~~
diff --git a/cmake/DARTCoreConfig.cmake.in b/cmake/DARTCoreConfig.cmake.in
index fcd6b80..ef699bb 100644
--- a/cmake/DARTCoreConfig.cmake.in
+++ b/cmake/DARTCoreConfig.cmake.in
@@ -11,12 +11,9 @@ get_filename_component(CURRENT_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
 set(DARTCore_INCLUDE_DIRS
   "@CMAKE_INSTALL_PREFIX@/include"
   "@Eigen_INCLUDE_DIRS@"
+  @BULLET_INCLUDE_DIRS@
 )
 
-if(HAVE_BULLET_COLLISION)
-  set(DARTCore_INCLUDE_DIRS ${DARTCore_INCLUDE_DIRS} "@BULLET_INCLUDE_DIRS@")
-endif()
-
 set(DARTCore_LIBRARY_DIRS "@CMAKE_INSTALL_PREFIX@/lib" "@Boost_LIBRARY_DIRS@")
 
 include("${CURRENT_DIR}/DARTCoreTargets.cmake")
~~~

scpeters, 2015-03-21:
I'll make a pull request for this.

jslee02, 2015-03-21:
Fixed by #361

PR #352 (2015-03-19): Add missing bullet includes
Fixes #351
PR #361 (2015-03-21): Fix BULLET_INCLUDE_DIRS in DARTCoreConfig.cmake
The fix from #352 doesn't work if the downstream code
doesn't have the HAVE_BULLET_COLLISION variable defined.
This change just inserts the BULLET_INCLUDE_DIRS
variable into DARTCore_INCLUDE_DIRS
and trusts that it will be blank and not affect
anything if it is unset.

Closes #351.
commit dbc3df4c (2015-03-21): Fix BULLET_INCLUDE_DIRS in DARTCoreConfig.cmake

The fix from #352 doesn't work if the downstream code
doesn't have the HAVE_BULLET_COLLISION variable defined.
This change just inserts the BULLET_INCLUDE_DIRS
variable into DARTCore_INCLUDE_DIRS
and trusts that it will be blank and not affect
anything if it is unset.

Closes #351.
closed 2015-03-21 by jslee02