dartsim/dart #358: Create alpha channel for shapes

mxgrey, 2015-03-18:
I'd like to make an alpha channel for visualization shapes. Is there a particular reason that colors in DART are limited to three components?

I can either implement the alpha channel as something independent of the three other channels, for example:
```
void Shape::setAlpha(double);
double Shape::getAlpha() const;
```

Or I could remake the color functions:
```
void Shape::setColor(const Eigen::Vector4d&);
const Eigen::Vector4d& Shape::getColor() const;
```

For setColor, this is not an issue, because we can just overload the function. getColor(), however, will not permit such simple overloading. Instead, we could have:
```
Eigen::Vector3d Shape::getColor() const;
const Eigen::Vector4d& Shape::getColor4() const;
```

So what I'm inclined to do have this API for enabling alpha:
```
void Shape::setColor(const Eigen::Vector3d&); // alpha gets set to 1.0
void Shape::setColor(const Eigen::Vector4d&);
Eigen::Vector3d Shape::getColor() const;
const Eigen::Vector4d& Shape::getColor4() const;
```

If people want, we could also have ``setAlpha()`` and ``getAlpha()`` functions.

jslee02, 2015-03-18:
> Is there a particular reason that colors in DART are limited to three components?

As I known, there is no. I think this feature is good to have.

> getColor(), however, will not permit such simple overloading. Instead, we could have:

We could use more concrete name like: 
```cpp
void Shape::setRGB(const Eigen::Vector3d&); // alpha gets set to 1.0
void Shape::setRGBA(const Eigen::Vector4d&);
const Eigen::Vector3d& Shape::getRGB() const;  // I'm not sure we can return segment of a Eigen vector as reference variable for getRGB().
const Eigen::Vector4d& Shape::getRGBA() const;
```

I'm also thinking that creating Color class that provides some useful features like:
* predefined color sets such as red, blue, and so on
* random color generation
* color operations

mxgrey, 2015-03-18:
That all sounds good to me. Although I'm inclined to still offer overloaded versions of setColor.

The only question I would have is ... what namespace would a Color class belong in? common? math? It could be defined in MathTypes, although it really has more to do with rendering/visualization than it has to do with math. It could possibly go in the renderer namespace.

In response to whether a ``const Eigen::Vector3d&`` return is possible, I'm pretty sure it's not. If we care about efficiency, we could return a Block type, but for the sake of API clarity, I think returning an Eigen::Vector3d by value will be just fine. Return value optimization will probably eliminate any difference in efficiency anyway.

mxgrey, 2015-03-22:
Finished by #359

closed 2015-03-22 by mxgrey