dartsim/dart #360: Consolidate file parsers

mxgrey, 2015-03-20:
I've found that it can be confusing to find the parser that's needed for a particular file, because of the way they are all specialized with different class names and different function names.

I think it would be good to offer a universal parser option. It would be a single class that can be utilized for any file type, and that uses the other parsers under the hood. It can either interpret the type of file based on file extension, internal file data, or a user-specified flag. The important thing is that it would be a single, unified API for all file parsing needs.

mxgrey, 2015-03-20:
Also, the parsing functions would be static members of the class, so you don't need to create an instance in order to do parsing.

jslee02, 2015-03-24:
Sounds good!

jslee02, 2015-06-30:
Started to work on this.

mxgrey, 2015-06-30:
I think it would also be good to consider #385 while making a universal parser. Right now, you need to know in advance whether a .skel file is meant to represent a single Skeleton or an entire World, and you need to ask the parser to return a Skeleton or a World specifically. I think it would be good to instead return a struct that contains ``std::vector<WorldPtr>``s and a ``std::vector<SkeletonPtr>``, and eventually also a ``std::vector<SimpleFramePtr>`` whenever we get around to adding SimpleFrames to the .skel format.

jslee02, 2015-06-30:
I got it. Once the current work (related to this issue) is wrapped up, I will create a pull request for it as a work-in-progress pull request, then add more changes for #385.

closed 2025-12-30 by jslee02