dartsim/dart #362: Installation test with CI

jslee02, 2015-03-21:
DART uses CI services of Travis CI and AppVeyor for build and unit tests but not installation.

scpeters, 2015-03-21:
~~~
diff --git a/ci/script_linux.sh b/ci/script_linux.sh
index 82eb3f3..283bfc3 100755
--- a/ci/script_linux.sh
+++ b/ci/script_linux.sh
@@ -1,6 +1,7 @@
  make
  sudo ldconfig --verbose # So the test executeables can detect libtinyxml2
  if [ $BUILD_CORE_ONLY = OFF ]; then make test; fi
+ sudo make install
  # cd tools/
  # ./code_check.sh
- # ./abi_check.sh 4.1.0  # Check with DART 4.1.0
\ No newline at end of file
+ # ./abi_check.sh 4.1.0  # Check with DART 4.1.0
diff --git a/ci/script_osx.sh b/ci/script_osx.sh
index 844bb27..12e416e 100755
--- a/ci/script_osx.sh
+++ b/ci/script_osx.sh
@@ -1,5 +1,6 @@
 make
 if [ $BUILD_CORE_ONLY = OFF ]; then make test; fi
+sudo make install
 # cd tools/
 # ./code_check.sh
-# ./abi_check.sh 4.1.0  # Check with DART 4.1.0
\ No newline at end of file
+# ./abi_check.sh 4.1.0  # Check with DART 4.1.0
~~~

jslee02, 2015-03-21:
Thanks! I [added additional script](https://github.com/dartsim/dart/commit/c97f29ca9280f8f0e33eef664c63669b8778a07b) that tries to build simple DART dependent application. Would this work to prevent errors like #351?

scpeters, 2015-03-21:
Yeah, that looks like it would help. We test example code that uses cmake `find_package` as well as `pkg_check_modules` (pkg-config).

jslee02, 2018-02-19:
Addressed by #998

PR #998 (2018-02-19): Add installation test to Travis
Resolves #362
closed 2018-02-19 by jslee02