dartsim/dart #368: Collision detectors should refresh collision shapes when they get changed

mxgrey, 2015-03-26:
Based on what I've looked through in the source code, it seems like the FCLCollisionDetector classes only grab the Shape information when a Skeleton is first given to it. This could pose a problem if a collision shape is ever added or removed (or even modified). It would be good if the fcl::CollisionGeometry data structures could be updated as Shapes change and added/removed as Shapes are added and removed from the BodyNode's collision shape array.

This could either be implemented with dirty flags or with the recent Signal/Slog feature that's been added.

Note that this does not appear to be an issue for the DARTCollide class.

jslee02, 2015-03-26:
Agree, and there is similar issue, #197.

mxgrey, 2015-04-28:
To add to this, the collision detectors should update whenever a new BodyNode is added to a Skeleton. I'm noticing that the collision handler behaves poorly when BodyNodes are created or removed from a Skeleton unless you remove the Skeleton and re-add it.

jslee02, 2016-04-16:
Related to #664

jslee02, 2021-04-22:
Resolved by #1112

closed 2021-04-22 by jslee02