dartsim/dart #373: Should PlaneShape(normal, point) really be deprecated?

mxgrey, 2015-04-09:
I noticed that the constructor has been deprecated in favor of PlaneShape(normal, offset), but is it really necessary to deprecate it? I can imagine it being convenient for an end user to construct a PlaneShape using normal and point information, and I don't see why having the two alternative constructors would hurt anything.

jslee02, 2015-04-14:
It's deprecated while changing the member variables from (normal, point(3d-vector)) to (normal, offset(scalar)) since point is redundant representation. I agree that `PlaneShape(normal, point)` is still convenient constructor. One minor concern is that we should provide `getPoint()` also or not.

mxgrey, 2015-04-15:
I think that the point information is sort of lost once the plane is constructed, because there are infinite points that can be used to define a plane. I don't really see that function as being necessary.

jslee02, 2015-04-15:
If the specific point information, which is provided the user, is fine to be lost, it looks good to have the constructor. I'll revert the deprecation.

mxgrey, 2015-04-30:
Looks like this has been resolved.

closed 2015-04-30 by mxgrey