dartsim/dart #379: Deprecated mark for clang

jslee02, 2015-04-24:
[DART doesn't define deprecated mark for clang](https://github.com/dartsim/dart/blob/d6e38a79a4f798354f6435ae03b87d92f79eff12/dart/common/Deprecated.h#L46).

mxgrey, 2015-05-11:
C++14 will introduce an ISO ``[[deprecated]]`` attribute tag, although I have no idea when that will be supported by GCC, Clang, and VS.

mkoval, 2015-05-11:
Until then, it like [clang implements the same extension as GCC](http://clang.llvm.org/docs/LanguageExtensions.html#messages-on-deprecated-and-unavailable-attributes):

```c++
  #define DEPRECATED(version) __attribute__ ((deprecated))
```

jslee02, 2015-05-12:
`[[deprecated]]` would be a simple and portable solution when all the three compilers support it.

Meanwhile, let's use the same extension as GCC. I'll create a pull request for this.

PR #384 (2015-05-13): Add deprecated and force-inline definitions for clang
resolves #379.
closed 2015-05-13 by jslee02