dartsim/dart #382: Remove public constructors for release 5.0?

mxgrey, 2015-05-04:
The life cycle for deprecation has been discussed in the past, and the general idea is things should be marked as deprecated for a while before they're completely removed.

However, for release 5.0, we want Skeletons to be managed strictly by shared_ptrs, which also means that Skeletons need to have knowledge of the shared_ptr that is managing them, so they can pass that shared_ptr to their BodyNodes. We want BodyNodes to be aware of the shared_ptr of the Skeleton that owns them so that users can obtain the shared_ptr of a Skeleton from its BodyNode.

To facilitate this, we want to create a static function called ``Skeleton::create()`` which will pass back a shared_ptr<Skeleton> where the Skeleton knows the value of its own shared_ptr. This means that constructing a Skeleton using its usual public constructor would be bad, because then the Skeleton would not have knowledge of its own shared_ptr, so the BodyNodes won't have the shared_ptr either.

Altogether, this means that keeping the public constructor for Skeleton would be harmful.

Taking this a step further, I wonder if we should keep the public constructors for BodyNode or Joint available at all. We could allow them to be marked as deprecated for a while before removing them entirely, but I'm worried that might be too gentle.

jslee02, 2015-05-05:
Yes, we have discussed. Basically, we try to follow [Semantic Versioning](http://semver.org/), and it's recommends on deprecated API is:
> Before you completely remove the functionality in a new major release there should be at least one minor release that contains the deprecation so that users can smoothly transition to the new API.

So it would be good to release dart 4.4 with deprecated API then dart 5.0 without them if it's possible. How do you think?

mkoval, 2015-05-05:
There is going to be an API breaking change in DART 5.0 regardless of this because `Skeleton::create` will change from returning a `Skeleton *` instead of a `shared_ptr<Skeleton>`. I'm not sure how much value there is in deprecating the `Skeleton` constructor if this is going to break anyway.

mxgrey, 2015-05-05:
> So it would be good to release dart 4.4 with deprecated API then dart 5.0 without them if it's possible.

We can do this the BodyNode and Joint constructors. I've been pretty careful so far to make it so that everything still works if you try to manually construct Skeletons in the old way (i.e. create a BodyNode, create a Joint, ``BodyNode::setParentJoint()``, ``BodyNode::addChildBodyNode()``, ``Skeleton::addBodyNode()``, ``Skeleton::init()``). However, creating a Skeleton on the stack would fundamentally break the code, because the mechanics of the new data structures assume that the Skeleton is managed by a shared_ptr.

mxgrey, 2015-05-05:
Then again, if we were to go with the [proposed group_ptr concept](https://github.com/mkoval/group_ptr/blob/master/group_ptr_test.cpp#L75), then the old BodyNode and Joint construction would break the code just as badly as the old Skeleton construction will.

jslee02, 2015-05-06:
Right, DART 5.0 will break API anyways, and actually that's why we bump up the major version number in this release. I just wonder if there is a point of releasing DART 4.4 with deprecated API as recommended in the [Semantic Versioning](http://semver.org/). But it seems not because old style construction of Skeleton, BodyNode, and Joint would break the code badly.

So +1 to removing public constructors in DART 5.0.

mxgrey, 2015-05-06:
We *could* release a version where BodyNode and Joint construction is deprecated, and it wouldn't break the code as long as the user is initializing their Skeletons correctly. But we can't release a version where the Skeleton's public constructor is deprecated, because publicly constructing a Skeleton *will* break the code badly.

I'm just not sure how valuable it is to release a minor version where one half of the API is broken but not the other half.

mxgrey, 2015-05-27:
The deprecated constructors have been removed in #369

closed 2015-05-27 by mxgrey