dartsim/dart #383: Proposal: Merge Skeleton::setPositions() and Skeleton::setPositionSegment()

mxgrey, 2015-05-11:
Rather than having two differently named functions for essentially the same purpose, I think it would make sense to just have one overloaded function:

```
void Skeleton::setPositions(const Eigen::VectorXd& _positions);
void Skeleton::setPositions(const Eigen::VectorXd& _positions, const std::vector<size_t>& _indices)
```

This proposal also applies to all the other functions that have a "Segment" version, such as ``Skeleton::setVelocitySegment()`` and ``Skeleton::setAccelerationSegment()``.

jslee02, 2015-05-11:
Sounds good to me.

mxgrey, 2015-05-11:
Actually, to match the existing pattern of ``setPosition(size_t, double)``, I'm thinking we should make it ``setPositions(const std::vector<size_t>&, const Eigen::VectorXd&)``.

jslee02, 2015-05-12:
Matching pattern looks good as well. How about applying this change to `getPositionSegment()`?

mxgrey, 2015-05-12:
Yep, I intend to do the same with every ``get[X]Segment`` function as well.

jslee02, 2015-05-12:
Great!

mxgrey, 2015-05-15:
This issue has been resolved in the memory_management branch #369

closed 2015-05-15 by mxgrey