dartsim/dart #387: SimWindow only renders one kinematic subtree per Skeleton

mkoval, 2015-05-21:
I have a `Skeleton` that has two root `BodyNode`s. Only the kinematic subtree attached to the first root, i.e. `skeleton.getRootBodyNode(0)`, is rendered by `SimWindow`. I suspect that something in `SimWindow` assumes that each `Skeleton` has only one root `BodyNode`, but I haven't tracked down the culprit.

I constructed the body by  by separately loading two URDF files with `DARTLoader`, then concatenated them with the recently added `moveTo` method. I can use `SimWindow` to render each of the bodies in isolation, so I don't think this is a problem with my URDF files.

mxgrey, 2015-05-21:
This is not surprising. I should have realized it beforehand. I can fix this with the same commit as #388

mxgrey, 2015-05-21:
This should be fixed now in #369

mkoval, 2015-05-24:
Thanks! This seems to be fixed now.

closed 2015-05-24 by mkoval