dartsim/dart #395: GlutWindow::screenshot() doesn't work

jslee02, 2015-05-26:


olzhas, 2016-02-17:
it works for me, but I have a question,

is it possible to change in /dart/gui/Win3D.cpp

``` C++
  if (mCapture)
    screenshot();
```

to

``` C++
  if (mCapture)
    std::async(std::launch::async, screenshot());
```

mxgrey, 2016-02-17:
Is there a particular motivation for launching it in a thread? I'd be concerned about race conditions causing problems, such as missing the frame that the user wanted, or worse: capturing a frame which is only partially rendered.

jslee02, 2016-02-17:
Out of curious, which platform are you using? I've never seen this function working on ubuntu.

olzhas, 2016-03-01:
I am using ubuntu as well, I did not check whether it is working on my computer, just saw it while reading more about C++11.

jslee02, 2016-03-01:
I meant `GlutWindow::screenshot()` function. I had observed that `GlutWindow::screenshot()` function doesn't work on my ubuntu (maybe other platforms as well). The console says "wrote screenshot frames/CaptureeXXXX.png`, but the files are actually not created at all. So I wondered if it really worked for you.

olzhas, 2016-03-01:
I guess it works on my computer. When I press 'c', `frames/CaptureXXXX.png` are getting generated.

jslee02, 2016-03-01:
I manually created `frames` directory, and not it works! It seems `lodepng` doesn't create necessary directories for the passed-in filename. `GlutWindow::screenshot()` should be changed to create `frames` directory when it doesn't exist. Thanks!

Also, I have a similar concern about the race conditions as @mxgrey mentioned. The screen data will be shared by the renderer and the screen-capturing thread.

olzhas, 2016-03-02:
which fix will be better

`char directory[8] = "frames";
  if (!boost::filesystem::exists(directory)) {
    boost::filesystem::create_directory(directory);
  }

  if (!boost::filesystem::is_directory(directory)) {
    std::cout << "frames/ is not a directory, cannot write a screenshot\n";
    return false;
  }`

jslee02, 2016-03-02:
I created a pull request for this (#623), which is similar to your suggestion but not using `boost::filesystem` as we're trying to minimize the dependencies as much as possible.

PR #623 (2016-03-03): Create 'frames' directory if not exists for screenshot
This fixes #395 thanks to @olzhas.
closed 2016-03-03 by jslee02