dartsim/dart #398: Example applications

mxgrey, 2015-05-27:
With all the new features coming up in the 5.0 release, it would seem appropriate to create some example applications that showcase the new DART features and how to use them.

The questions I have are: (1) What are some good toy problems for showcasing the features, and (2) What format should we use for the examples?

With respect to (1), it would be good to discuss what kinds of problems people are interested in, and what kinds of things people want to use DART for. I think there are some obvious simple things we can do:

- Parse a World from a skel file and load it into a simulation. The file will contain a triple pendulum, and the user can apply torques on the pendulum's joints by pressing the 1, 2, and 3 keys. The value of the torque can be adjusted with Q, W, E, A, S, D (Q/A will increase/decrease torque 1, W/S will increase/decrease torque 2, and E/D will increase/decrease torque 3). The color of each BodyNode in the pendulum will change based on its torque level.
  - This would demonstrate how to set the generalized forces during simulation and how to change the coloring of Shapes
- Parse three Skeletons from URDF and/or SDF files, and load them into a simulation. 
  - While the simulation is running, the user can press 1, 2, or 3 to clone one of the original three Skeletons and place it at some random location in the air, from which it will fall. (There will be a floor for everything to land on)
  - The user can also press Q, W, or E to remove a random copy of the Skeleton corresponding to 1, 2, or 3.
  - If two of the Skeletons have SoftBodyNodes and one of them is rigid, this could make for some interesting collision interactions
  - This would demonstrate how cloning and memory management works
- Construct a tree-structured Skeleton manually and show it simulating. Allow the user to apply torques during simulation using the keys 1-9, similar to the first example.

I have more ideas for examples that might be interesting, but many of them would require more interaction than what we can get with the current SimWindow. Once I have the OSG features finished, I'll make more examples using that.

If anyone has ideas for more examples, or features they'd like to see in action, let me know. In particular, I think we should do something to demonstrate the use of Frame semantics, but I haven't thought of a good example yet.

Now regarding question (2) above, the way I would like to do this would be something like the ``examples`` folder [that I have in my OSG branch](https://github.com/dartsim/dart/tree/grey/osg/osgDart/examples). Each .cpp file is an individual and self-contained example. This is in contrast to the current ``apps`` directory in DART where each app has its own folder. I think putting all the code into a single file would be less confusing for beginners, and I don't think any of these examples would require very much code anyway.

karenliu, 2015-05-28:
This is also a good opportunity for us to revisit all the old examples we have. Some are still good but need to be modified for DART 5.0 framework. Before we come up with a list of examples, let's first figure out the table of contents of our tutorial document. Once we know what topics we want to cover, it's easier to generate examples ideas for each topic.

karenliu, 2015-05-28:
This is a straw man proposal for the tutorial topics and examples. It includes what we currently have in dart-tutotiral.pdf and new features in later versions of DART.

1. Three-link pendulum:
- show basic simulation under gravity
- show how to change gravity coefficient, change time step
- add perturbation forces via UI: change color of body node when force is applied; draw arrow to indicate force
- change joint properties: rest pose, stiffness, damping, joint limits
- show explicit vs implicit joint dynamics
- show how to save/load generalized state
- show how to modify skeleton: add two more links with different joint types
- create closed loop constraint

2. Three-link mixed skeleton falling on the floor
- basic simulation of a skeleton with one soft cylinder node, one rigid ellipsoid node, and one box node with protruding rigid part
- change the parameters of soft body: surface deformation, rigid deformation, damping
- change contact friction coefficient
- change to different collision detectors (FCL/Bullet)

3. Manipulator + object
- load a URDF file: KR5 sixx R650
- load a world with a floor and a table 
- add/delete objects (rectangle dominos)
- simple IK to reach the first domino
- show how to create PD servo to track the IK pose
- show operational space control and use it to push the first domino: show how to access M, C, \dot{J}, J
- add a weld joint between the manipulator and one of the object and let the user to move the object via UI

4. Biped 
- show how to access various Jacobian matrices (linear, angular, wrt COM, wrt point, etc)
- show how to add gravity compensation
- show how to use hybrid dynamics for controller design
- demo “frame" semantics