dartsim/dart #399: Better support for mesh assets in .skel files

mxgrey, 2015-05-27:
[I've noticed here](https://github.com/dartsim/dart/blob/c127038b91914f1a65d4e0b877cf053ce3ff8184/dart/utils/SkelParser.cpp#L841) that the SkelParser assumes all mesh file names will be given with respect to DART_DATA_PATH. This is obviously extremely fragile.

What would be a better way to define mesh file locations? I imagine it should be relative to the directory that the original .skel file is in. The most robust way to do it would be the ``package:`` style of ROS, but that seems like overkill to me.

mxgrey, 2015-05-27:
Also, it looks like SDF might not be very robust. It assumes that the URI for a mesh is nothing more than the mesh's file name with respect to the Skeleton's path. I don't know the SDF convention offhand, but I'm guessing the URI might have something like package information in it, similar to URDF.

jslee02, 2025-11-21:
Closing as we're deprecating the SKEL format