dartsim/dart #400: Compiling Problem on Fedora 22

thunderbirdtr, 2015-05-29:
Hello, I have problem with compiling and my cmake result is ; 

-- The C compiler identification is GNU 5.1.1
-- The CXX compiler identification is GNU 5.1.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- ============================================
--                 DART 4.3.4
-- ============================================
-- 
-- [ Mandatory dependencies ]
-- Looking for Eigen - found
-- Looking for FCL - found
-- Looking for Assimp - found
-- Looking for OpenGL - found
-- Looking for GLUT - found
-- Looking for Boost - 1.57.0 found
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Looking for FLANN - found
-- Looking for TinyXML - found
-- Looking for TinyXML2 - found
-- Looking for urdfdom - found
-- 
-- [ Optional dependencies ]
-- Looking for OpenMP - found
-- Looking for NLOPT - found
-- Looking for IPOPT - found
-- Looking for SHARK - found
-- Looking for BulletCollision - found
-- Looking for PerlModules - found
-- Looking for Doxygen - found
-- 
-- [ Build summary ]
-- CMAKE_GENERATOR  : Unix Makefiles
-- Compiler ID      : GNU
-- Compiler version : 5.1.1

-- Build type       : Release
-- BUILD_SHARED_LIBS: ON
-- ENABLE_OPENMP    : ON
-- Build core only  : OFF
-- Build examples   : ON
-- Build unit tests : ON
-- Install path     : /usr/local
-- CXX_FLAGS        : -Wall -msse2 -fPIC -std=c++11
-- CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_SOURCE_DIR : /home/thunderbirdtr/ros_build/dart
-- CMAKE_BINARY_DIR : /home/thunderbirdtr/ros_build/dart/build
-- 
-- [ Configured files ]
-- /home/thunderbirdtr/ros_build/dart/build/dart/config.h
-- /home/thunderbirdtr/ros_build/dart/build/cmake/DARTCoreConfig.cmake
-- /home/thunderbirdtr/ros_build/dart/build/cmake/DARTConfigVersion.cmake
-- /home/thunderbirdtr/ros_build/dart/build/cmake/DARTConfig.cmake
-- /home/thunderbirdtr/ros_build/dart/build/cmake/DARTConfigVersion.cmake
-- /home/thunderbirdtr/ros_build/dart/build/cmake/dart.pc
CMake Warning (dev) at CMakeLists.txt:453 (configure_file):
  configure_file called with unknown argument(s):

   @only

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at cmake/DARTMacros.cmake:76 (message):
  Installing deprecated Constraint.h.  This should be removed in DART 5.0.
Call Stack (most recent call first):
  dart/constraint/CMakeLists.txt:40 (dt_issue_303)

CMake Warning at cmake/DARTMacros.cmake:76 (message):
  Installing deprecated LCPSolver.h.  This should be removed in DART 5.0.
Call Stack (most recent call first):
  dart/lcpsolver/CMakeLists.txt:35 (dt_issue_303)

-- 
-- [ Unit tests ]
-- Adding test: testCollision
-- Adding test: testNameManagement
-- Adding test: testForwardKinematics
-- Adding test: testBuilding
-- Adding test: testGeometry
-- Adding test: testNearestNeighbor
-- Adding test: testParser
-- Adding test: testJoints
-- Adding test: testCommon
-- Adding test: testMath
-- Adding test: testOptimizer
-- Adding test: testDynamics
-- Adding test: testInverseKinematics
-- Adding test: testUtilities
-- Adding test: testWorld
-- Adding test: testSoftDynamics
-- Adding test: testConstraint
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/thunderbirdtr/ros_build/dart/build

I want to integrate to gazebo 5 but when I try to compile it ; 

/home/thunderbirdtr/ros_build/dart/dart/dynamics/Skeleton.h:67:8: note: declared here
 struct GenCoordInfo
        ^
/home/thunderbirdtr/ros_build/dart/dart/dynamics/Skeleton.h:716:15: warning: 'GenCoordInfo' is deprecated [-Wdeprecated-declarations]
   std::vector<GenCoordInfo> mGenCoordInfos;
               ^
/home/thunderbirdtr/ros_build/dart/dart/dynamics/Skeleton.h:67:8: note: declared here
 struct GenCoordInfo
        ^
In file included from /usr/include/bullet/LinearMath/btVector3.h:21:0,
                 from /usr/include/bullet/BulletCollision/CollisionDispatch/btCollisionWorld.h:80,
                 from /usr/include/bullet/btBulletCollisionCommon.h:22,
                 from /home/thunderbirdtr/ros_build/dart/dart/collision/bullet/BulletCollisionDetector.h:43,
                 from /home/thunderbirdtr/ros_build/dart/dart/utils/SkelParser.cpp:46:
/usr/include/bullet/LinearMath/btScalar.h:337:14: warning: 'btInfinityMask' defined but not used [-Wunused-variable]
  static  int btInfinityMask = 0x7F800000;
              ^
Reaping winning child 0x232c770 PID 30554 
Removing child 0x232c770 PID 30554 from chain.
Reaping losing child 0x245ef10 PID 30539 
CMakeFiles/Makefile2:208: recipe for target 'dart/CMakeFiles/dart.dir/all' failed
make[1]: *** [dart/CMakeFiles/dart.dir/all] Error 2
Removing child 0x245ef10 PID 30539 from chain.
Reaping losing child 0x2028010 PID 30517 
Makefile:146: recipe for target 'all' failed
make: *** [all] Error 2
Removing child 0x2028010 PID 30517 from chain.

This is what I got so far, how I handle this ? 
Thank you.

mxgrey, 2015-05-29:
It looks like the error is related to Bullet. Bullet is an optional dependency for DART, and to be honest I've never tried compiling while I had Bullet installed, so I wouldn't be able to guess why it might be failing.

A cheap and temporary workaround (just to check whether Bullet is the real culprit) would be to comment out (or delete) [lines 279 - 286 of the root CMakeLists.txt](https://github.com/dartsim/dart/blob/d1ec56ce35e9b70d63e312560bfd5b1bbe071a30/CMakeLists.txt#L279). If removing those lines allows DART to compile, then at least we'll have narrowed the issue down to being Bullet related. My best guess would be something to do with incompatible library versions.

jslee02, 2015-05-29:
I changed the style of the post for better readability.

It seems not the whole compile error log, and doesn't include the error complains (two warnings). It might not be related to Bullet. Could you post the whole build log?

mxgrey, 2015-05-29:
Ah yeah, the Bullet message is just a warning about an unused variable. There has to be something else saying what actually failed.

thunderbirdtr, 2015-05-29:
make[2]: *** [dart/CMakeFiles/dart.dir/optimizer/ipopt/IpoptSolver.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [dart/CMakeFiles/dart.dir/all] Error 2
make: *** [all] Error 2

I can't copy to here so I uploaded to "gist" I hope It will be help

https://gist.githubusercontent.com/thunderbirdtr/38cd0cdd89d9011c23ba/raw/dart_error

mxgrey, 2015-05-29:
The only place I see it mention an error is with IPOPT:

```
[1;30mIn file included from /usr/include/coin/IpUtils.hpp[0m:13:0,
                 from [1;30m/usr/include/coin/IpTNLP.hpp[0m:[35m12[0m,
                 from [1;30m/home/thunderbirdtr/ros_build/dart/dart/optimizer/ipopt/IpoptSolver.h[0m:[35m45[0m,
                 from [1;30m/home/thunderbirdtr/ros_build/dart/dart/optimizer/ipopt/IpoptSolver.cpp[0m:[35m37[0m:
[1;30m/usr/include/coin/IpTypes.hpp[0m:[35m26:9[0m: [1;31merror[0m: [31m'[36mFORTRAN_INTEGER_TYPE[0m[31m' does not name a type[0m
 typedef FORTRAN_INTEGER_TYPE ipfint;
         ^
```

I would guess that it's a versioning issue. IPOPT is an optional dependency, so you could try commenting out (or deleting) [line 255 in the root CMakeLitst.txt](https://github.com/dartsim/dart/blob/d1ec56ce35e9b70d63e312560bfd5b1bbe071a30/CMakeLists.txt#L255). If removing that line allows DART to compile, then we'll need to update the IPOPT extension and/or specify the correct version in CMake.

thunderbirdtr, 2015-05-29:
Problem solved I deleted ipopt from my system and compile process finished. So yes you need to specify correct version of it. 

So I have this version of "Ipopt"

```
 coin-or-Ipopt                       x86_64         3.12.3-1.fc22                  fedora          776 k
 coin-or-Ipopt-common                noarch         3.12.3-1.fc22                  fedora          6.9 M
 coin-or-Ipopt-devel                 x86_64         3.12.3-1.fc22                  fedora          216 k
 coin-or-Ipopt-openmpi               x86_64         3.12.3-1.fc22                  fedora          776 k
 coin-or-Ipopt-openmpi-devel         x86_64         3.12.3-1.fc22                  fedora          216 k
```

jslee02, 2017-11-12:
I believe this is resolved. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02