dartsim/dart #411: Windows Issues with current Master build

jturner65, 2015-06-17:
I've mentioned these in an email to JS but I will list them here as well.  Didn't think these warranted their own issues, but if so, I apologize.

1) NOMINMAX :
could all the #include <windows.h> be replaced with something along the following?
(from http://stackoverflow.com/questions/4913922/possible-problems-with-nominmax-on-visual-c/4914108#4914108 )
	#ifdef NOMINMAX
	#include <windows.h>
	#else
	#define NOMINMAX
	#include <windows.h>
	#undef NOMINMAX
	#endif

2) "small" variable name :
another <windows.h> issue (sigh)

http://stackoverflow.com/questions/27793470/why-does-smallgive-an-error-about-char
could the variable named "small" in Win3D.cpp line 77 be changed to something else?  windows.h includes a file which in turn #define's "small"

3) GL_MULTISAMPLE
http://stackoverflow.com/questions/4207506/where-is-gl-multisample-defined
this macro is not defined in windows, and is apparently outdated.

jslee02, 2015-06-17:
1) and 2) are resolved in [windows_build_issues branch](https://github.com/dartsim/dart/tree/windows_build_issues).

I learned that GL_MULTISAMPLE is supported since OpenGL 1.3 from [here](https://www.opengl.org/wiki/History_of_OpenGL#OpenGL_1.3_.282001.29) but I couldn't find any document says it's outdated. Can you point out any reference that says which version of OpenGL is used in windows and if GL_MULTISAMPLE is supported? It seems there are many changes since OpenGL 3.0 but I still stay in older version. :S

jturner65, 2015-06-17:
sure.  here’s a SO link that discusses it :
http://stackoverflow.com/questions/4207506/where-is-gl-multisample-defined

i guess a conditional redefinition of it could solve the problem.

From: Jeongseok Lee 
Sent: Wednesday, June 17, 2015 12:02 PM
To: dartsim/dart 
Cc: John Turner 
Subject: Re: [dart] Windows Issues with current Master build (#411)

1) and 2) are resolved in windows_build_issues branch.

I learned that GL_MULTISAMPLE is supported since OpenGL 1.3 from here but I couldn't find any document says it's outdated. Can you point out any reference that says which version of OpenGL is used in windows and if GL_MULTISAMPLE is supported? It seems there are many changes since OpenGL 3.0 but I still stay in older version. :S

mkoval, 2015-06-17:
@jturner65 That post actually says exactly the opposite (in one of the comments):

> Umm... That should read "used to be an extension", multisampling's been a part of OpenGL core for a long time now.

In any case, there will need to be a workaround to make it work on Windows. Regardless of what the standard says. :smile:

jturner65, 2015-06-17:
i think you meant to quote JS.  :)

mxgrey, 2015-06-17:
It looks like the SO thread is saying that GL_MULTISAMPLE *is* part of the OpenGL specifications (not outdated), but was not included in Windows for mysterious reasons.

mkoval, 2015-06-17:
@jturner65 That is a quote from the StackOverflow thread you linked to. See @mxgrey's reply.

jturner65, 2015-06-17:
mea culpa, i misunderstood the implication of "used to be an extension".  regardless, it doesn't seem to be available in windows. maybe GL_MULTISAMPLE_ARB

jturner65, 2015-06-25:
how do we want to address this issue?  GL_MULTISAMPLE_ARB is also not natively supported in windows.  perhaps we can include a more recent version of gl.h in golems install (the one natively installed by windows has a most recent copyright date of 1996).

jslee02, 2015-08-01:
Created a [pull request](https://github.com/dartsim/dart/pull/482) for NOMINMAX and "small" variable name issues.

OpenGL and GLUT compatibility issues will be handled once we move on [GLFW](http://www.glfw.org/)

closed 2015-10-16 by jturner65