dartsim/dart #413: SkelParser.cpp eigen issues

jturner65, 2015-06-17:
The new SkelParser.cpp file makes use of many stl containers (specifically maps) of structures containing vectorizable Eigen constructs.  as per this page : 

http://eigen.tuxfamily.org/dox/group__TopicStlContainers.html

these maps need to specify alignment allocators.

furthermore, Joint.h : struct Properties requires  EIGEN_MAKE_ALIGNED_OPERATOR_NEW  (I'm pretty sure an Isometry3d is vectorizable and so subject to this requirement)

jturner65, 2015-06-17:
Also, i am having an issue with the "return std::make_shared<...>" code.  I have found that boost::make_shared will not work with objects requiring alignment, and i think that's the issue here as well.  see this thread : 

https://forum.kde.org/viewtopic.php?f=74&t=97936

specifically this  : 
"you need to pass this allocator:
Eigen::aligned_allocator<Foo_d>
to allocate_shared(...); "

mxgrey, 2015-06-17:
Just out of curiosity, @jturner65 , what platform are you getting these issues on?

We were negligent about explicitly dealing with Eigen's alignment problems, because we thought this was resolved with C++11, but it seems that not all platforms support that part of the standard yet. [It looks like Windows is one such culprit.](https://msdn.microsoft.com/en-us/library/hh567368.aspx)

mkoval, 2015-06-17:
@mxgrey I've seen many more alignment issues on `x86` than `x86_64`.

@jturner65 Are you running 32-bit or 64-bit Windows?

jturner65, 2015-06-17:
i am on windows 7 64, using visual studio 2013 32-bit.  

with regard to the alignment issues, i wouldn't think any standard adherence would help this since, as i understand it, Eigen's alignment requirements are related to SIMD optimisations.

mxgrey, 2015-06-17:
We haven't been experiencing any alignment-related issues with GCC and Clang, which both have the [C++11 alignment support](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2007/n2341.pdf) that VS2013 only has "partial" support for.

[You can see on this page](http://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html) that according to the Eigen developers, the alignment issues are resolved in C++11:

> No, it's not our bug. It's more like an inherent problem of the C++98 language specification, and seems to be taken care of in the upcoming language revision: [see this document](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2007/n2341.pdf).

jturner65, 2015-06-17:
cool, thanks for the link - it's good to know that at least some compilers don't have this issue.

mxgrey, 2015-06-17:
Hopefully once Microsoft catches up, the whole issue can be a thing of the past. We thought it already was, but I guess we need to get some active Windows developers on the team.

jturner65, 2015-06-17:
i'd love to help :- ) i've been living in dart for a couple years now.  i'll help out any way i can.

jturner65, 2015-06-25:
fixed with #414

closed 2015-06-25 by jturner65