dartsim/dart #418: Build error with libbullet

olzhas, 2015-06-24:
/usr/local/include/dart/collision/bullet/BulletTypes.h:50: error: 'btInfinityMask' was not declared in this scopec
   return btInfinityMask;
          ^

The problem can be fixed removing the line BulletTypes.h:50.

jslee02, 2015-06-26:
The dummy function was added to suppress unused-variable-warnings due to btInfinityMask but it doesn't seem to be exposed on all the platforms. Actually [bitInfinityMask is surrounded by some preprocessors](https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btScalar.h#L364), and Bullet 2.83 added [similar code](https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btScalar.h#L366) only when the variable is exposed.

I don't know how to expose the dummy function only when bitInfinityMask is exposed. Maybe we should remove the dummy function and let the compiler to complains warnings.

jslee02, 2015-08-01:
This should be fixed by #480. Please confirm.

jslee02, 2015-08-02:
Closing for now. Please reopen if it sill doesn't work.

closed 2015-08-02 by jslee02