dartsim/dart #420: CMake Issue with windows - setting static vs dynamic

jturner65, 2015-06-25:
at line 46 of CMakeLists.txt in the main dart repo, static is specified as the default for MSVC, but this doesn't work (i think this is because of the golems build we have available).  i have changed :
```
if(MSVC)
  set(DART_RUNTIME_LIBRARY "/MT" CACHE STRING "BaseName chosen by the user at CMake configure time")
... 
```
to  :   
```
if(MSVC)
  set(DART_RUNTIME_LIBRARY "/MD" CACHE STRING "BaseName chosen by the user at CMake configure time")
 
```
 in my own project and it seems to work ok, although if there is a better option, please let me know.

jslee02, 2015-06-26:
What do you mean by "doesn't work"? If you use DART as other project's dependency, then they should use same option; use /MT or /MD for both projects.

mxgrey, 2015-06-26:
I'm guessing he means that DART's dependencies as provided by the "golems" installer have a flag that's incompatible with the flag in DART's cmake.

jslee02, 2015-06-26:
I see. Then we can change the default flag to /MD for DART. 

Do we need to provide the dependencies for /MT as well? I wish Windows also have package management system like Linux and OSX. There are [nuget](https://www.nuget.org/) and [chocolate](https://chocolatey.org/) but they seem to have none (or few) packages of DART's dependencies yet.

jturner65, 2015-06-26:
yes, mxgrey is correct. apologies for my lack of clarity.  
couldn't this be set as a cmake option/checkbox to go between MD and MT?  

also, when i TA'ed 4496 i noticed that there were built both static and dynamic versions of the golems prereqs - maybe those could be provided again?

jturner65, 2015-06-26:
i just checked, and nugget's support isn't too bad, from what i could find.  there's only a 3.0 package for assimp, and no packages for fcl, flann, console_bridge and urdfdom, but libccd, Eigen, glut, googletest and tinyxml2 seem to be supported and up to date.

it probably wouldn't be too difficult to make nugget packages for those that are lacking them.  might be a better option than the "golemsinstall.exe" route.

closed 2016-10-19 by jturner65