dartsim/dart #424: Problem with angular momentum conservation in dart 4.3.4

scpeters, 2015-06-26:
Initially reported as [gazebo issue 1487](https://bitbucket.org/osrf/gazebo/issue/1487/dart-has-lower-accuracy-on-angular). We added a [GetWorldAngularMomentum test](https://bitbucket.org/osrf/gazebo/src/3e1e86e456efae91240b61a42238e437ab07bdad/test/integration/physics_link.cc#cl-241), but we had to [relax the tolerances for dart](https://bitbucket.org/osrf/gazebo/src/3e1e86e456efae91240b61a42238e437ab07bdad/test/integration/physics_link.cc#cl-310). This may be related to #229; I'm not sure. Also, I haven't run the test with dart 5.

jslee02, 2015-10-15:
This issue should be resolved by #518. I will check the gazebo test again and also add an test to DART to make sure this is kept being resolved.

scpeters, 2015-10-15:
I believe #518 improved angular momentum conservation, but the linear velocity and position seem to be much less accurate now.

scpeters, 2015-10-15:
My last comment is based on a test using https://github.com/scpeters/benchmark

acxz, 2019-02-17:
I was just going through DART issues and was wondering if we could close this issue, since it is on 4.3.4 (an outdated version).
The issue could be kept if the problem persists in the current stable branch of DART tho.

mxgrey, 2019-02-18:
We should probably have a regression test for the issue and see that the regression test passes before we consider closing it.

jslee02, 2019-02-18:
As a note, probably this is related to https://github.com/dartsim/dart/issues/1193.

jslee02, 2019-07-14:
Closing this issue in favor of #1193, which I'm still working on.

closed 2019-07-14 by jslee02