dartsim/dart #433: OMPL integration

olzhas, 2015-07-03:
Hello everyone,
I found that DART provides only one motion planning algorithm (RRT). There is OMPL (open motion planning library) that provides implementation of a state-of-art motion and path planning algorithms. I worked with it for some time. I can try to integrate with DART, if you guide me. I meant could you tell me what should it look like and I will code it, because I am not quite familiar with design patterns. 
Also I found that there was a discussion regarding OMPL and DART integration (http://wiki.ros.org/fuerte/Planning/Simulation%20and%20Control/JointMeeting), maybe you know what was the outcome?

mkoval, 2015-07-03:
@psigen and I wrote DART bindings for OMPL. It consists of two parts: (1) an OMPL `StateSpace` that wraps a set of `DegreeOfFreedom`s and (2) an OMPL `StateValidityChecker` that internally uses DART for collision checking. This is sufficient to call any of OMPL's geometric motion planners on a DART environment.

We don't currently have this code available in a usable state, but we are in the process of writing documentation and cleaning up the build system. I'll post a pointer to it once it's online.

mxgrey, 2015-07-03:
Seems like this can be marked as closed since the work is in progress, right? Of course we can reopen it if people want to continue discussing the matter.

jslee02, 2015-07-03:
How about leave this open and mark "in-progress" so that people know this is on going?

mxgrey, 2015-07-03:
Makes sense.

mkoval, 2015-07-07:
The current version of our interface is [available in this repository](https://github.com/personalrobotics/r3). We weren't planning on posting this publicly for a while, but I did my best to update the `README` with the information you need to build it.

There are a few caveats:

1. We haven't rigorously tested this code yet and, in particular, the `StateValidityChecker`.
2. Any `DegreeOfFreedom` with unbounded limits (i.e. +/- infinity) is assumed to be continuous revolution joint and map them to an `SO2StateSpace`. This means that the DOF's values will wrap after a full revolution. This will break if you have another joint with unbounded limits (e.g. a `FreeJoint`).
3. Only geometric motion planning is supported.

Please do file an issue on the repository if you run into trouble. We'll do our best to help.

mxgrey, 2015-07-07:
Just out of curiosity, is the limit wrapping issue because of the way we need to interface to OMPL? I'm assuming it's because RRT- and PRM-style geometric motion planners require finite joint limits, so OMPL is demanding finite joint limits.

We could probably have an OMPL-ize function that has slightly more sane default limits for each joint type, and assigns those default limits to unbounded joints based on their type.

psigen, 2015-07-07:
The limit wrapping issue is because we have no mechanism for differentiating between configuration spaces that wrap (such as continuously rotating joints) and ones that are simply infinite (such as the unbounded free joint).  In OpenRAVE, for example, there is a special IsCircular flag and some settings to set the wrapping bounds per axis.

The reason this needs to be different is because the topological connectivity of an unbounded rotational configuration space is not the same as an unbounded one.  As a simple 1DOF example, if you have a PRM node at a position `2*PI` and one at `0.0001`, if they are in the configuration space of continuously rotating joint, they should be connected, but if they are X position of an unbounded gantry, they should not.

Note that in the most general case the actual wrapping can vary as well: e.g. if you are in a situation where the configuration space is representing something other than angles, such as a pre-transmission motor position or a joint DOF that controls a linkage, or if you are measuring angular position in encoder ticks or some other unit.

@mkoval can probably explain in more precise terms if needed.

mkoval, 2015-07-07:
@psigen is correct. The key issue is that the topology of the space is different: SO(2) instead of R.

I don't think we need to support other types of wrapping. Fundamentally, the only way to introduce an SO2-like topology into the system is to add a continuous rotation joint. In your two examples, they are behind the transmission and in the linkage, respectively.

Transmissions have implications beyond joint wrapping. E.g. power transmission, actuator limits, etc. I don't think DART differentiates between actuators and joints (@mxgrey, can you confirm?), so this would be a large change.

**Proposal:** Add a `isCircular` function to `DegreeOfFreedom` and `Joint`. This function will return `true` is the `DegreeOfFreedom` has an SO(2) topology and `false` if it does not.

mxgrey, 2015-07-07:
You are correct that DART does not differentiate between joints and actuators. DART essentially pretends that actuators are magical black boxes, and we only care about the final joint torque that comes at the end of the transmission.

The continuously rotational SO(2) thing is an interesting matter that I hadn't really considered. Most of the revolute joints that I've worked with are not able to continuously revolve because of internal wiring, so even if they can rotate more than 360-degrees, they are not able to "wrap to pi" like a continuous revolute joint.

I suppose an ``isCircular`` flag would make sense. We could give the RevoluteJoint a property for whether or not it's circular while the other Joint types are just known ahead of time. I think BallJoint and the rotational portion of FreeJoint will naturally wrap themselves because they use log maps, so I'm guessing they would just always qualify as circular?

psigen, 2015-07-07:
I believe that's correct behavior.

@mkoval: Is there a better name than `IsCircular`?  I know we had considered `IsContinuous` at one point.

mkoval, 2015-07-07:
> Most of the revolute joints that I've worked with are not able to continuously revolve because of internal wiring, so even if they can rotate more than 360-degrees, they are not able to "wrap to pi" like a continuous revolute joint.

Yes, I've only seen it on robots that use sliprings internally to internally route cables. One example is Kinova JACO and MICO arms.

> I suppose an isCircular function would make sense. We could give the RevoluteJoint a property for whether or not it's circular while the other Joint types are just known ahead of time.

A `RevoluteJoint` is circular iff it has no joint limits (like you mentioned). I don't think it's necessary to add an extra flag here. Instead, we would implement something like:
```c++
bool RevoluteJoint::isCircular() const
{
  return std::isinf(mLowerLimit) && std::isinf(mUpperLimit);
}
```

> I think BallJoint and the rotational portion of FreeJoint will naturally wrap themselves because they use log maps, so I'm guessing they would just always qualify as circular?

Yes, that's correct. I think there are a few others, too:

- `RevoluteJoint`
- `UniversalJoint`
- `BallJoint`
- `PlanarJoint` rotation axis only
- `FreeJoint` rotation axis only

In all cases, we should only return `true` of the respective axis has no limits.

@mxgrey @psigen Does that sound correct?

mxgrey, 2015-07-07:
That all sounds correct and good.

I might prefer a better name than ``isCircular``, though, if we can think of one. ``isContinuous`` seems like it might be better, except that I usually associate the word "continuous" as being the alternative to "discrete". I'll try to brainstorm an alternative word, but I might not come up with anything.

psigen, 2015-07-07:
Oh that's true @mkoval, I hadn't thought about the redundancy of joint limits for the specific case of revolute joints.  The rest sounds fine.

mkoval, 2015-07-08:
After a quick glance, URDF uses `continuous`, OpenRAVE uses `circular`, and V-REP uses `cyclic`.

I don't like `continuous` for exactly the reason @mxgrey pointed out: it sounds too much like the opposite of `discrete`. I'm fine with `circular` or `cyclic` (or something better, if you come up with it).

mxgrey, 2015-07-08:
Of all the options so far, I think my vote would be with ``cyclic``. It's very accurate and unambiguous.

mkoval, 2015-07-08:
I took a stab at implementing this in #441. Please take a look and let me know what you think.

psigen, 2015-07-08:
Thanks @mkoval!

Not to conflate issues, but would it make sense to throw in an `isLimited()` that just checks the `std::isinf(mLowerLimit) && std::isinf(mUpperLimit)` on any joint axis, and then use that for the `isCyclic()` check where appropriate?

Just curious if it's a useful helper function on its own for translation axes in addition to providing a little code reuse here (could be implemented on the base Joint class).

mxgrey, 2015-07-08:
We'll want to call it ``isPositionLimited()``, but yeah, that sounds to me like it would be good. We could also have ``isVelocityLimited()``, ``isAccelerationLimited()``, and ``isForceLimited()``.

mkoval, 2015-07-08:
@psigen That's a good idea. Randomized motion planning only works if the measure of the space is bounded, so it's a check we'll need elsewhere.

jslee02, 2018-02-19:
DART + OMPL is now available on top of [AIKIDO](https://github.com/personalrobotics/aikido) (see https://github.com/personalrobotics/aikido/pull/51 and [directory](https://github.com/personalrobotics/aikido/tree/master/include/aikido/planner/ompl)). AIKIDO project is still under development, but it's actively improving so please let us know if you find any difficulties.

closed 2018-02-19 by jslee02