dartsim/dart #434: TinyXML vs TinyXML2

mkoval, 2015-07-06:
DART currently depends on both TinyXML (version 2.6.2) and TinyXML2. I initially thought that these two packages were the same, but they actually do correspond to separate Debian packages (`libtinyxml-dev` vs. `libtinyxml2-dev`). Compilation breaks if I remove the dependency on TinyXML2.

What is the difference between these two versions? Can we remove one of these dependencies?

mxgrey, 2015-07-06:
It looks like the URDF parser (which is the ``urdfdom`` library) depends on tinyxml, while DART's native skel parser depends on tinyxml2.

I'm all in favor of minimizing dependencies, but at the same time I'd rather not use a worse/older version of a library. It doesn't look like we can get rid of tinyxml if we want to be able to use ``urdfdom``, so either we'd have to adopt the old tinyxml library everywhere or we'd have to write our own version of ``urdfdom``, which would also involve a complete rewrite of the urdf parser. Unless someone knows of a urdf parsing alternative to ``urdfdom``.

mkoval, 2015-07-07:
I see. If TinyXML is a transitive dependency of `urdfdom`, then DART shouldn't have to depend on it explicitly (it should be handled by `find_package(urdfdom)`. Does DART itself use TinyXML (not TinyXML2) anywhere internally?

mxgrey, 2015-07-07:
I can't say for sure, but if we *are* using both TinyXML and TinyXML2 internally, then we should definitely consolidate on one of them, preferably TinyXML2.

mxgrey, 2015-07-07:
I did a comprehensive search, and it turns out that our interface to urdfdom does use TinyXML, because urdfdom seems to take in a ``TiXmlDocument`` to parse.

I know our urdf parser implementation is very old, so it might be possible to avoid the TinyXML interface. I'll take a quick look at this. But if the API of urdfdom hasn't changed, then we're stuck with a TinyXML dependency.

mxgrey, 2015-07-07:
Okay, I need to revise my statement. ``urdfdom`` does **not** take in a ``TiXmlDocument``.

The situation is this: ``urdfdom`` takes in a ``std::string`` and passes back a ``ModelInterface`` which represents a single robot. This means that ``urdfdom`` is not suitable for handling a urdf that contains more than one model/robot/skeleton. We therefore have a layer of translation that goes through an xml document, finds each model, and passes them one by one into ``urdfdom``. Our layer of translation was written using TinyXML, for reasons that were probably lost three years ago.

It seems like we should be able to replace our use of TinyXML with TinyXML2 in that layer of translation. Although before doing that, we might want to see if there's a version of ``urdfdom`` that actually supports parsing in multiple models from a single file. Our urdf loader was actually written around the same time that ``urdfdom`` was first being released, so there might be more features available now.

mxgrey, 2015-07-07:
The urdfdom API on Github seems to have not changed since our implementation of the urdf parser roughly three years ago, so it looks like we'll need to just switch from TinyXML to TinyXML2 in our loader.

ana-GT, 2015-07-09:
As far as I remember, one of the reasons why we decided to keep using the urdf parser was because back then Gazebo used the same package for reading urdfs). Is this still the case?

jslee02, 2016-01-31:
Yes. [urdfdom package is still used to parse urdf files by sdformat](https://bitbucket.org/osrf/sdformat/src/2f84aac3eeb5d6f42fc3f1504ef4c2e5081f52a1/src/parser_urdf.cc?at=default&fileviewer=file-view-default#parser_urdf.cc-27), which is a stand alone package separated from Gazebo. 

> The urdfdom API on Github seems to have not changed since our implementation of the urdf parser roughly three years ago, so it looks like we'll need to just switch from TinyXML to TinyXML2 in our loader.

:+1: As urdf takes in a `std::string`, we can use any xml parser to parsing in urdf file for the layer of translation.

jslee02, 2017-03-21:
> This means that urdfdom is not suitable for handling a urdf that contains more than one model/robot/skeleton.

Hm, it seems urdf is supposed to contain a robot per file according to the [website](http://wiki.ros.org/urdf/XML/model). Do we need to care the case that urdf contains multiple robots?

mxgrey, 2017-03-21:
It was mostly for our own convenience. I think for a while, URDF was the sole file format used by DART (besides the skel format, which inconveniently needs to be manually written), but most people wanted to be able to store entire worlds within individual files. We customized the URDF format a little bit to accommodate this.

I believe that removing this feature would break backwards compatibility for numerous users, and I don't believe it has any negative impact besides the xml parsing library dependency. But if we want to remove the TinyXML dependency, we could just switch to using TinyXML2 for that portion of the library, and I don't believe it would have any negative impact.

jslee02, 2017-03-21:
I didn't intend to remove the functionality as long as it's functional. I just wanted to make sure if the <world> element is included in the spec or a custom feature. Thank you for the clarification.

One minor confusion is that urdf has [`urdf::World`](https://github.com/ros/urdfdom_headers/blob/5cb0b6b6a99eeeff5cd111dabd0093a29305a2f9/urdf_world/include/urdf_world/world.h#L93) class with a [example world XML](https://github.com/ros/urdfdom_headers/blob/5cb0b6b6a99eeeff5cd111dabd0093a29305a2f9/urdf_world/include/urdf_world/world.h#L41) in the repo but not on the [website](http://ros.org/wiki/usdf/XML/urdf_world). Maybe the extension was planned but [hasn't finished](https://github.com/ros/urdfdom/blob/master/urdf_parser/src/world.cpp#L52) yet.

For now, I think it would be nice to (1) have at least one urdf that contains multiple models with a simple test for it and (2) explicitly mention that the functionality is extended from the original spec. I could do this but don't think it's an urgent issue though.

mxgrey, 2017-03-21:
> Maybe the extension was planned but hasn't finished yet.

Yes, we (specifically @ana-GT if I recall correctly) inquired about having multiple robots in a single URDF, and the urdfdom maintainers kindly provided some unfinished/unofficial code for us to work off of.

> For now, I think it would be nice to (1) have at least one urdf that contains multiple models with a simple test for it and (2) explicitly mention that the functionality is extended from the original spec. I could do this but don't think it's an urgent issue though.

Those both sound like very good ideas.

jslee02, 2018-02-13:
This issue can be closed once #858 is merged.

jslee02, 2018-02-18:
Addressed by #993

PR #993 (2018-02-18): Remove tinyxml dependency
This PR is a revision of #858.

As discovered in #434, `tinyxml` is only a transitive dependency of `urdfdom` and DART doesn't use it in the public API. This PR drops the explicit dependency of `tinyxml`.

Resolves #434 

***

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
closed 2018-02-18 by jslee02