dartsim/dart #436: Bizarre behavior with DART, potential stability bug

Skylion007, 2015-07-07:
I am using DART to simulate laminated robotics. Something really odd occurs though when two bodies are colliding. It almost appears as if there is a massive amount of friction that is keeping the body in place, but alas I cannot figure out how to specify friction or lower said value. I was wondering why this [odd behaviour](https://www.youtube.com/watch?v=5lZN0qZ2mzM) occurs.

I am using Gazebo 1.4 and specifying the values of the model via an SDF. I then articulate one of the joints to push off the ground, but somehow the other joint stays fixed to the ground almost like a vaulting pole, I am wondering what I am doing wrong in this case. Are there any values I am specifying incorrectly? Or is this a bug with how Gazebo handles contact? I can provide the SDF if that would help in any way, but I think the video explains the situation pretty well.

mxgrey, 2015-07-07:
Let me know if I'm interpreting things correctly: When you push off the ground with the foot on the right side of the robot, the left foot stays planted on the ground, and the robot rotates around it like a pivot. Then, it eventually ends up folded up on itself while remaining unnaturally upright.

Is this behavior symmetric? If you push off the ground with the left foot, will it behave the same way?

You could try tweaking things like friction coefficients. I think the default coefficient of friction is 1.0, which is the highest amount of friction that is physically plausible for most kinds of objects. High friction makes for more numerically stable simulations, which I think is why we make the default so large. But friction shouldn't prevent the thing from tipping over.

Other than that, inertia values like mass, center of mass, and moment of inertia have a huge impact on the realism of a simulation, and I could easily imagine a mistake in one of those causing this kind of behavior, so you might want to double-check that those are getting entered correctly. I'm not sure how exactly you would do that in Gazebo.

If you want to send me the SDF and let me know what kind of joint forces you're applying, I can run it in the latest version of DART (outside of Gazebo) and see how it behaves. I can also play around with the inertial parameters to see if that might be causing it.

Skylion007, 2015-07-07:
Yes, the behavior is symmetric.

I haven't implemented inertia tensors yet, but center of mass, and mass are both correct. I've deleted the necessary files to make the SDF, but I'll put them together and send them to you soon. I am also going to file a ticket with Gazebo to allow you to set the friction in their next release of SDF.

Skylion007, 2015-07-09:
I have recreated an SDF that exhibits the same behaviour (sorry I overwrote the original SDF will making changes to my exporter).

```
<sdf version="1.5">
  <world name="world">
    <model name="default_body">
      <static>false</static>
      <pose>0 0 0 0 0 0</pose>
      <link name="140494037398544">
        <gravity>true</gravity>
        <self_collide>true</self_collide>
        <visual name="basic_bot_visual0-0">
          <pose>-0.309982060197 0.24753164557 0.0 0 0 0</pose>
          <geometry>
            <polyline>
              <point>0.351632936588 -0.25</point>
              <point>-0.344490079445 -0.25</point>
              <point>-0.344490079445 0.25</point>
              <point>0.337347222303 0.25</point>
              <height>2.5e-08</height>
            </polyline>
          </geometry>
          <cast_shadows>true</cast_shadows>
        </visual>
        <visual name="basic_bot_visual0-1">
          <pose>-0.309982060197 0.24753164557 2.5e-05 0 0 0</pose>
          <geometry>
            <polyline>
              <point>0.351632936588 -0.25</point>
              <point>-0.344490079445 -0.25</point>
              <point>-0.344490079445 0.25</point>
              <point>0.337347222303 0.25</point>
              <height>2.5e-08</height>
            </polyline>
          </geometry>
          <cast_shadows>true</cast_shadows>
        </visual>
        <visual name="basic_bot_visual0-2">
          <pose>-0.304985021837 0.24753164557 5e-05 0 0 0</pose>
          <geometry>
            <polyline>
              <point>0.356629974948 -0.25</point>
              <point>-0.349487117805 -0.25</point>
              <point>-0.349487117805 0.25</point>
              <point>0.342344260662 0.25</point>
              <height>2.5e-08</height>
            </polyline>
          </geometry>
          <cast_shadows>true</cast_shadows>
        </visual>
        <visual name="basic_bot_visual0-3">
          <pose>-0.309982060197 0.24753164557 7.5e-05 0 0 0</pose>
          <geometry>
            <polyline>
              <point>0.351632936588 -0.25</point>
              <point>-0.344490079445 -0.25</point>
              <point>-0.344490079445 0.25</point>
              <point>0.337347222303 0.25</point>
              <height>2.5e-08</height>
            </polyline>
          </geometry>
          <cast_shadows>true</cast_shadows>
        </visual>
        <visual name="basic_bot_visual0-4">
          <pose>-0.309982060197 0.24753164557 0.0001 0 0 0</pose>
          <geometry>
            <polyline>
              <point>0.351632936588 -0.25</point>
              <point>-0.344490079445 -0.25</point>
              <point>-0.344490079445 0.25</point>
[... 545 lines cut ...]
        <child>140494037305680</child>
        <pose>0.804545722727 0.18000006 5e-05 0 0 0</pose>
        <axis>
          <xyz>0.00684915441279 0.999976544267 0</xyz>
          <limit>
            <lower>-3.145159</lower>
            <upper>3.14519</upper>
          </limit>
          <dynamics>
            <damping>1000.0</damping>

            <spring_stiffness>1000.0</spring_stiffness>
          </dynamics>
        </axis>
      </joint>
    </model>
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
  </world>
</sdf>
```

mxgrey, 2015-07-09:
Cool. If you also let me know what kind of forces you were applying, I can test out the behavior on my end.

Skylion007, 2015-07-09:
I was originally applying torque at one of the joint, but I have replaced that with springs to make the simulation self-contained.

mxgrey, 2015-07-09:
Your SDF appears to use the PolyLine tag, which is supported in Gazebo but not supported natively in DART (yet). This is able to work in Gazebo because Gazebo has built-in utilities that convert the PolyLine shapes into regular meshes.

Would it be possible to generate a version of the SDF that uses box primitives or meshes instead? Otherwise, it may be a while before I can test out your situation, because I'd have a lot of workarounds that I'd need to set up.

Skylion007, 2015-07-09:
Here, I used my extrusion algorithm to generate meshes and here is the resulting [sdf and meshes](http://www.mediafire.com/download/5w45way5c5754l8/data.zip)

mxgrey, 2015-07-10:
I've loaded up your model and poked around with it a bit. Here are my observations:

The your bodies' masses are *shockingly* small. They're so small that right from the get-go they're going to be suffering from issues with numerical precision and numerical stability. By that I mean: no matter what kind of dynamics algorithm you use, if you're using a digital computer you're going to run into problems related to rounding errors and unstable inversions with such small values for your masses.

Another thing is that without specifying your bodies' moments of inertia, you're going to be left with the simulation equivalent of ``undefined behavior``. When you load a file using DART's native SDF loader, if you neglect to specify the moments of inertia then the loader will take a feeble guess at what the moment of inertia tensor might be based on the mesh geometry and the mass that's provided. There is absolutely no promise that this guess will be even remotely reliable, especially for meshes with unusual geometry. I don't know if Gazebo does something similar to this, but no matter what, you're taking a gamble when you don't specify the moments of inertia.

The first thing I would recommend doing is checking if your material density is correct. The estimate that I'm computing for your bodies' densities using DART are around ``4.88e-6 mg/cm^3``. For comparison, the cutoff for the [ultralight material category](https://en.wikipedia.org/wiki/Ultralight_material) is **10** mg/cm^3. A super high tech material called [Aerographene](https://en.wikipedia.org/wiki/Aerographene) has a density of ``0.16 mg/cm^3``, and it is considered the world's lightest solid material. Now I won't pretend that my estimated computation of your bodies' densities are accurate , but I doubt they'd be off by six orders of magnitude.

Skylion007, 2015-07-10:
Hmm, I am working on inertial densities. I think you are right about my density calculations, I appear to have a conversion bug in my density calculation algorithm. It should be about 1 gram per cm^3. I'll get right on fixing that. However, in my earlier simulations I floor it to 0.01g per link and still got some bizarre behaviour so I would just check to make sure that Dart can handle centigrams. Thanks so much for finding that bug though!