dartsim/dart #448: Signal when a MetaSkeleton's structure changes

mkoval, 2015-07-09:
I'm writing some code that relies on the indexing of a `MetaSkeleton`. I'd like to invalidate my calculations if the structure of the skeleton changes; i.e. a `BodyNode` is added or removed, a `Joint` changes type, etc. I thought this was possible using `Signal`s, but I can't seem to find it.

Is this possible in the current API?

mxgrey, 2015-07-09:
We can certainly put in a signal that gets triggered when indexing changes, but I don't believe that exists yet.

mxgrey, 2016-04-14:
Now that we have the ``VersionCounter`` class, we could create a signal that gets triggered each time an object's version is incremented. Subscribing a slot to a Skeleton's version signal will then activate the slot each time the structural properties change. Alternatively, you can track a Skeleton's version number and do whatever handling is needed whenever you see a new version.

mkoval, 2016-04-15:
> Alternatively, you can track a Skeleton's version number and do whatever handling is needed whenever you see a new version.

:+1: This is actually a better than a signal for my purposes. I am closing this issue.

closed 2016-04-15 by mkoval