dartsim/dart #450: Dart doesn't like heightmaps? What am I doing wrong?

Skylion007, 2015-07-09:
While I was using Gazebo, I am posting this here because it seems like a Dart specific problem. Whenever I specify the Dart engine in Gazebo, I am unable to use height maps. Here is the SDF file, I can provide the troublesome terrain image if that is causing issues as well

[Here](http://www.mediafire.com/download/d9f3ci7gkumejxd/heightmap.zip) are two worlds files that both crash when opened in Dart, but do not when opened in ODE or Simbody.

mxgrey, 2015-07-09:
DART, in fact, does not support heightmaps. Of course it would be preferable if it would print a warning and not crash, but that would have to be done on the Gazebo end of things.

mxgrey, 2015-07-09:
[The source code for Gazebo](https://bitbucket.org/osrf/gazebo/src/5bae2bdbd0d32b055764e9f11fd2c179008c0d31/gazebo/physics/dart/DARTHeightmapShape.cc?at=default) seems to suggest that the expected behavior is to print a warning and do nothing, but I would guess that maybe this non-initialized shape instance is being used recklessly somewhere which results in a crash.

jslee02, 2015-07-09:
Unfortunately, DART doesn't support heightmaps yet either in [Gazebo](https://bitbucket.org/osrf/gazebo/issues/909/dart-doesnt-support-meshes-or-heightmaps) and DART itself. 

The main reason was that DART's default collision detector, [FCL](https://github.com/flexible-collision-library/fcl), doesn't support heightmap feature. I'm not sure if FCL has specific plan for heightmap support. In the meantime, I will test ODE and Bullet to see if they are suitable for DART supporting all features DART requires including heightmaps. I'll update more details once the tests are done.

> Of course it would be preferable if it would print a warning and not crash, but that would have to be done on the Gazebo end of things.

Right, I'm actually working on gazebo code right now. I'll make it to print a warning instead of crash.

jslee02, 2018-08-24:
I believe this is resolved by #1069.

closed 2018-08-24 by jslee02