dartsim/dart #465: Does DART have support for loop constraints?

Skylion007, 2015-07-22:
I am having some minor stability problems with this model: 
[image]
I am curious if DART has support for loop constraints. If it does not, it could definitely be causing my problem. Otherwise, feel free to comment and close this issue.

mxgrey, 2015-07-22:
If you mean closed kinematic chains, there is some limited support for them.

Kinematic chains can be closed by applying a joint-like constraint between two bodies. Currently there are two joint types that can be used for this: Ball Joint (which behaves like a ball-and-socket joint) and Weld Joint (which just rigidly attaches the two bodies).

However, I don't know if these features are accessible through Gazebo.

Skylion007, 2015-07-22:
Ah, so it does not have support for Kinematic Chains via Hinge Joints?

mxgrey, 2015-07-22:
Correct, not currently.

However, it may be possible to do a workaround where you have a dummy body attached by a normal hinge (revolute) joint and then close the loop by attaching the dummy body to the intended attachment with a weld joint.

The only catch with this is that using the weld joint to close a kinematic chain can result in numerical stability if there are not enough linearly independent degrees of freedom in the chain. For example, if you have:

**Body1** -> *RevoluteJoint* -> **Body2** -> *RevoluteJoint* -> **Body3** -> *RevoluteJoint* -> **DummyBody** -> *WeldJointConstraint* -> **Body1** 

Then this may end up being numerically unstable and produce ``NaN`` results (at least in my experience). In your case, it *might* be okay since the robot's root joint is completely unconstrained, but I haven't tested that scenario, so I don't know for sure.

Skylion007, 2015-07-22:
If this method does work, coding this into DART to hack support for this would be a nice feature. It appears to know when there is a loop joint because it prints out stuff like this:

The collision detector already has a collision node for body node [139789387640400].
The collision detector already has a collision node for body node [139789389371920].
The collision detector already has a collision node for body node [139789389506256].
The collision detector already has a collision node for body node [139789643254736].
The collision detector already has a collision node for body node [139789387180176].
The collision detector already has a collision node for body node [139789386954064].

mkoval, 2015-07-22:
The right solution would be to add a constraint-based revolute joint to DART, rather than adding native support for the workaround that @mxgrey described. I haven't worked with the dynamics simulation code very much, but I believe that this is relatively straightforward.

Skylion007, 2015-07-22:
How straightforward? Like two weeks straightforward or six month straight forward? While I might be able to hack something together for individual meshes, I am working on a generic model exporter and adding that hack into the export procedure, but that would prove difficult especially since Gazebo does not give you access to a Weld Joint Constraint by default. I fear this might force me to switch physics engines.

mxgrey, 2015-07-22:
I don't know of any DART developers who are currently working with anything related to the dynamics of skeletons with closed kinematic chains, so I wouldn't expect it to be implemented any time soon unless someone graciously volunteers.

It's important to remember that all of DART's ongoing development is done by graduate students who all have their own research obligations, and who just contribute to DART whenever their time permits or their interests align. So feature requests aren't likely to be fulfilled unless it's (1) completely trivial or (2) sufficiently relevant to the interests or needs of a contributor. I don't expect that this feature request would qualify as trivial, although I invite someone to prove me wrong.

But absolutely anyone is welcome to submit pull requests, so if someone wants to implement the other types of closed loop constraints, then we will be glad to review the pull request and merge it in if it's satisfactory.

jslee02, 2015-07-23:
As @mxgrey said, closed loop constraint can be made by (1) adding dummy BodyNode (or breaking existing BodyNode in the closed loop into two BodyNode) using WeldJointConstraint or (2) using BallJointConstraint manually in DART but [it's not supported through Gazebo](https://bitbucket.org/osrf/gazebo/issues/913/dart-does-not-detect-closed-loop) yet.

What I can say now is: I have seven major items to improve DART within 6 months, and auto-constructing closed loop constraint is in the list.

danaukes, 2015-07-23:
@jslee02: That would absolutely be useful.  @mxgrey, I get it; we're all in the same boat.  Would it be easier to implement generic closed-loop constraints, or to expose the weld joint type to gazebo?

Thanks!

mxgrey, 2018-12-03:
Gazebo uses DART as one of four options that you can choose for Gazebo's underlying physics engine. Gazebo itself is a wrapper that leverages a physics engine to provide a full-fledged simulation program (simulating physics and sensors, providing rendering, a GUI, saving/loading worlds, recording data, etc).

Different versions of Gazebo target different versions of DART. For example, I think Gazebo version 7-8 target version 4 of DART, while Gazebo 9 targets version 6 of DART. The DART versions that Gazebo uses are no different than the DART versions that you'll find here.

Gazebo's API is designed to support closed kinematic chains of all types, but DART only provides "native" support for using weld joints and ball joints as closed kinematic chains. To overcome this limitation, if you load a closed kinematic chain while Gazebo is using DART, Gazebo will automatically split one link in the chain, divide the inertial properties between the two halves of the link, and then rejoin the two halves with a weld joint constraint. This workaround allows the Gazebo API to support all types of closed kinematic chains, even though the native DART API only supports weld joints and ball joints.

closed 2025-11-24 by jslee02