dartsim/dart #471: Add Frame::getTransform(withRespecTo, inCoordinatesOf)

jslee02, 2015-07-26:
We can specify relative frame and observation frame when we get velocities and accelerations of a `Frame`, but observation frame is not allow to specify for the transformation.

mxgrey, 2015-07-26:
The main reason that I didn't add this along with the others is... I guess I'm not quite sure what the meaning or use of it would be. Would we just apply a rotation to the transform based on the coordinates of the ``inCoordinatesOf`` frame? I'd be okay with this API addition, I've just never personally known a use for it.

jslee02, 2015-07-27:
> Would we just apply a rotation to the transform based on the coordinates of the inCoordinatesOf frame?

Yes. We may want to know a relative transformation between two bodies in a different body like observing the relative transformation using a sensor that is attached to a different body.

jslee02, 2015-08-02:
Added by #475.

closed 2015-08-02 by jslee02