dartsim/dart #476: find_package(DART) does not fail if DART is not found and REQUIRED is specified

mkoval, 2015-07-29:
I am including DART in my `CMakeLists.txt` file with:

```cmake
find_package(DART REQUIRED)
```

Due to an unrelated issue with my DART installation, this failed:

```
-- Could NOT find Dart (missing:  DART_ROOT) 
```

However, the `find_package` script did not raise an error so my build proceeded and failed because it could not find DART's headers.

mxgrey, 2015-07-29:
I'm a bit confused about what the ``DART_ROOT`` variable is. I know I've seen *environment* variables like ``BOOST_ROOT``, but I haven't seen ``<package>_ROOT`` as a CMake variable. With CMake, I'm used to seeing ``<package>_INCLUDE_DIRS`` and ``<package>_LIBRARY_DIRS``.

mkoval, 2015-07-29:
Actually, despite the_very confusing_ error message, this is not DART's fault at all. CMake found the [`Dart.cmake` file **included with CMake**](http://www.cmake.org/cmake/help/v3.2/module/Dart.html). This is not an issue on Linux because `Dart` is not equal to` DART`. However, Mac OS X defaults to using a case-insensitive filesystem...

closed 2015-07-29 by mkoval