dartsim/dart #477: CMake package conventions and conformity

mxgrey, 2015-07-29:
For the sake of pull request #461 I started digging through some of CMake's packaging documentation and also reviewing DART's current cmake setup. I thought I would discuss some of my findings and my thoughts on how we might improve DART's packaging. Some of this also comes from discussions with @mkoval.

* I've noticed that according to [CMake documentation](http://www.cmake.org/cmake/help/v3.0/command/link_directories.html), the libraries that are returned by ``find_package`` are supposed to be absolute paths, but DART currently just returns library names. It may be a good idea to follow the convention that's stated by the doc.

* The split between ``dart-core`` and ``dart`` is understandable, but I think it could be handled more gracefully. What I would propose is renaming ``dart-core`` to just be ``dart``, and then the extra features that are currently in ``dart`` should each be split out into its own ``COMPONENT`` based on their dependencies. For example, the contents of ``dart/gui`` which depend on OpenGL should go into a ``dart-gui`` library. The contents of ``dart/planning`` which depend on FLANN should go into a ``dart-planning`` library.

It gets a bit trickier when it comes to components that are nested within namespaces. For instance, ``dart/optimizer/nlopt`` could go into a library named ``dart-nlopt`` or it could go into ``dart-optimizer-nlopt``. The first option offers a shorter name, which is nice, but the second option is more precise and consistent. The second option has the added benefit of avoiding possible future collisions, for example if we end up with a ``dart/planner/nlopt`` extension which is independent of the ``dart/optimizer/nlopt``. After giving it some thought, I think I would prefer using the full name, but I'm not necessarily opposed to the shortened name.

* We would use the ``find_package(DART COMPONENTS xxx)`` feature of CMake to allow the user to decide what they want to include in the ``${DART_LIBRARIES}`` variable. For example, ``find_package(DART COMPONENTS gui planning utils-sdf)`` would set ``${DART_LIBRARIES}`` to include ``dart`` along with the GUI library, planning library, and SDF parsing library.

* Each of the libraries would be available on the apt repo independently of each other, giving the user more control over which dependencies they want to deal with.

I think this overall approach would conform better to how CMake packages are usually handled. It may also make DART easier for end users to install, because they'll have a finer level of control over the dependencies.

mxgrey, 2015-07-29:
It occurs to me that, while this may be easy from the cmake side of things, it could make the debian packaging significantly more complex (and maybe Homebrew as well, but I've never touched Homebrew before, so I don't know). I wonder if it would be possible/useful to have cmake auto-generate the debian packaging files using the ``configure_file`` command.

psigen, 2015-07-29:
One minor thought: If you did actually leave `dart-core` as `dart_core` (e.g. `libdart_core.so`), that might help eliminate the ambiguity against the existing `DART` CMake implementation that @mkoval experience recently.

Or really any naming convention that makes CMake use a name other than just `DART` when resolving this.  There are just so many projects named `DART` these days. :)

mxgrey, 2015-07-29:
This hasn't been mentioned here yet, but we're in the process of deciding upon an entirely new name for DART, so that conflict won't be an issue in the future. There should be an official announcement about this in the not-too-distant future. So everywhere that I used "dart" in the earlier posts, just imagine that's a placeholder for whatever the new library name will be.

scpeters, 2015-07-29:
The components of `dart_core` used to be called [rtql8](https://bitbucket.org/karenliu/rtql8), which I always thought was a clever, unique name.

jslee02, 2015-07-29:
`rtql8` is one of the candidates. One possible issue with `rtql8` is it includes number at the end so that it causes confusion with version number.

scpeters, 2015-07-29:
One random example of software with number in name: [x264](http://packages.ubuntu.com/trusty/x264) video encoder

psigen, 2015-07-29:
`x264` makes sense, since the `x264` codec is well-known specification. Maybe `rtqlate` would be a reasonable compromise?

My thought is that while the name is pretty cute, it's also nice to have something plain and readable, especially if it needs to get used in literature references or international codebases where the onomatopoeia isn't going to make any sense.

Consider cases like [libiberty](https://gcc.gnu.org/onlinedocs/libiberty/), which is named to get the cute linker flag `-liberty`, but then gets compiled and referred as the typo-friendly `libiberty.a`.

mxgrey, 2015-07-29:
We should probably hold off on discussing the merits of potential names until the official thread is posted (we'll be posting it as a Github issue). It'll be better to have all that discussion consolidated in one place.

jslee02, 2015-10-16:
Beside on the renaming of DART, I like the idea of reorganizing DART as proposed. This should be able to install dart more systematically with the package manage systems. For example, we cannot use BulletCollisionDetector when we install `dart-core` or `dart` from `apt-get`. In order to use it, we should build DART from the source on a system `libbullet-dev` is installed. However, we would be able to use it by installing `dart-collision-bullet` (it would of course require `libbullet-dev` to be installed).

If anyone have a plan on implementing this I would like to do.

```
-----------------------
libdart
-----------------------
```

psigen, 2015-10-16:
@mkoval and I were also discussing the merits of reorganizing into modules as far as contributing new functionality.  Packages can be built and tested in separate repositories, and later be merged as components (as they become mature) without heavy restructuring.

mxgrey, 2015-10-16:
If you're referring to git submodules, I've heard both very good and very bad things about using them. I haven't attempted to use them myself, so I don't know exactly where to stand on them. If anyone is experienced with using them and has good reference material for them, I'd be interested in learning more.

jslee02, 2015-10-16:
I also haven't used the functionality yet. It would be good chance to learn about it to me.

psigen, 2015-10-16:
I wasn't actually referring to git submodules... they have various tradeoffs in complexity.

I just meant that we could easily move code into the mainline repo without having to merge files into the source directories and refactor  the build system.

jslee02, 2015-10-16:
I don't have strong objection but am not sure if it is worth to have multiple repositories for each modules when the modules are quite small. It would be possible to resolve the `package.xml` issue for different packages discussed in #523 though.

It would be good if someone can let me know the advantages of doing so.

psigen, 2015-10-16:
In general, I would advise against many small repositories:
- It is much harder to version than one repository, all of the repositories need to be forked together for API changes.
- It means that

mkoval, 2015-10-16:
I agree with @psigen. I think that it is a great idea to split DART into multiple libraries to help manage the dependency explosion. However, I very much **do not** want DART split into multiple repositories for the reasons that he outlined.

Even though all of the modules are in one repository, I think each module should:
- have defined inter-module dependencies
- have defined external dependencies
- correspond to a `include` directory with the same name
- be exposed as a `COMPONENT` in CMake

My biggest confusion with multiple-component projects is the confusion about what you need to depend on. This is a constant frustration with Boost: the only way I know if I need to depend on `boost-system` is to run the build and see if linking fails. I don't want to have a repeat experience in DART. :smile:

jslee02, 2016-05-09:
I believe this is addressed by #587. Please reopen if there are more things need to be handled.

closed 2016-05-09 by jslee02