dartsim/dart #478: Missing symbols because of default destructor

mkoval, 2015-07-29:
I have to manually link my project  against Nlopt to use the DART Nlopt wrapper on OS X. If I do not, I get this linking error:
```
Undefined symbols for architecture x86_64:
  "_nlopt_destroy", referenced from:
      nlopt::opt::~opt() in MoveHandStraightVectorField.cpp.o
ld: symbol(s) not found for architecture x86_64
```
I think what is happening is that:

1. `NloptSolver` has a member variable of type `std::unique_ptr<nlopt::opt>`
2. The `NloptSolver` destructor is defined as `default`
3. Therefore, `NloptSolver::~NloptSolver` is instantiated in the user's code.
4. The `std::unique_ptr<nlopt::opt>` destructor is not called inside DART.
5. Therefore, the `std::unique_ptr<nlopt::opt>` is instantiated inside the user's code
6. The user's code does not have the `_nlopt_destroy` symbol available, so linking fails.

As expected, manually defining the default constructor in the `NloptSolver.cpp` file solved the issue. In theory, this can happen *any time* we use the `default` keyword. The only foolproof solution I can think of is to stop using the `default` keyword entirely. :frowning: 

I discovered this because the OS X linker is *extremely* picky and will not use symbols that are included as part of a transitive dependency. I suspect that it could happen on other platforms, too.

mxgrey, 2015-07-29:
That's really unfortunate behavior. I'll make the changes to the ``grey/osg`` branch.

mxgrey, 2015-07-29:
As for never using the ``default`` keyword, that really only seems relevant for classes that use an external dependency. I don't see a reason to not use it in pure DART classes.

mxgrey, 2015-07-30:
I've pushed the fix for this to #461. If it's still not working correctly, let me know.

mkoval, 2015-08-02:
:+1: Fixed in #461.

I suspect that there are other instances of this in DART, but we'll just have to squash them when they pop up.

closed 2015-08-02 by mkoval