dartsim/dart #489: Segfault from Linkage::Criteria::expandToTarget()

jslee02, 2015-08-16:
[Linkage.cpp Ln:364](https://github.com/dartsim/dart/blob/master/dart/dynamics/Linkage.cpp#L365)
```cpp
  // If we have successfully reached the target, expand from there
  if(newBns.back() == _target.mNode.lock())
    expansionPolicy(_target.mNode.lock(), _target.mPolicy, newBns);
```

Segfault is raised at `newBns.back()` when newBns is empty. According to [cppreference](http://en.cppreference.com/w/cpp/container/vector/back), calling back on an empty container is undefined. It happens on Windows but not on Ubuntu or Mac.

mxgrey, 2015-08-16:
This is actually a really easy fix. This is just an unhandled edge case where the linkage wasn't able to expand towards the target whatsoever. The current code incorrectly assumes that it was able to take at least one step towards the target. We just need to change the if-statement to ``if(newBns.size() > 0 && newBns.back() == _target.mNode.lock())``. I'll make a pull request with the fix.

jslee02, 2015-08-16:
Thanks! I confirmed that it fixes the segfault. I just made a comment on #490 about minor code preference.

jslee02, 2015-08-17:
Fixed by #490, #491, and #492.

closed 2015-08-17 by jslee02