dartsim/dart #524: Joint has the default copy constructor

mkoval, 2015-10-15:
I noticed that `Joint` does not `= delete` its copy constructor. I believe that all of the fields in `Joint` are copyable, so this means that the default copy (and, possibly, move) constructor are being implicitly defined by the compiler.

I suspect that is not intentional (and potentially dangerous!) because every `Joint` is supposed to be owned by it's child `BodyNode`.

mxgrey, 2015-10-15:
It's definitely not intentional, and it's definitely dangerous. I'll submit a pull request to correct this later today.

mxgrey, 2015-10-15:
It's also worth checking if this is an issue for BodyNode and Skeleton, since none of those should permit a copy construction.

mkoval, 2015-10-16:
@mxgrey fixed this in #525. Thanks!

mkoval, 2015-10-23:
`MultiDofJoint` and `ZeroDofJoint` still have copy constructors. :smile:

mxgrey, 2015-10-23:
I'm pretty sure they'd technically be deleted by default since they inherit from a class whose copy constructor is deleted. Still, it would be good to be explicit about it.

mkoval, 2015-10-23:
They're not - I discovered this because I'm using `libclang` to parse the DART source code and they're listed in the output. :laughing:

mkoval, 2015-10-23:
`DegreeOfFreedom` also has a copy constructor.

jslee02, 2015-10-25:
This seems to be resolved by #539.

closed 2015-10-25 by jslee02