dartsim/dart #545: Getting properties of a bounding box.

mklingen, 2015-11-03:
In `Shape`, there is a function called `getBoundingBoxDim`. This is pretty ambiguous because a bounding box has *both* a center and dimensions. There are two ways of storing this: either with a center + dims, or with a min + max. How do I get the properties of the bounding box when all I have are the dimensions? Also, are these half-extents or side lengths?

mklingen, 2015-11-03:
I see there is already a `TODO` for this

https://github.com/dartsim/dart/blob/59f72f916e3e8197f5c18c77f6c37c2333a80b19/dart/dynamics/Shape.h#L120

mxgrey, 2015-11-03:
The ``Shape`` class API is overdue for some renovation. We're planning some pretty big changes for the next major release. Any suggestions for changes to the ``Shape`` API are extremely welcome right now. Since ``Shape`` won't be backwards compatible after the next major release anyway, now is a really good time to be making any significant API changes.

mklingen, 2015-11-03:
I've started a branch to fix the bounding box issues. Will post it here when its in a good state. I will be switching things to `min/max` semantics and keeping around a `BoundingBox` structure that can compute all the other needed stuff from that.

mklingen, 2015-11-03:
https://github.com/dartsim/dart/pull/547

closed 2015-12-01 by mklingen