dartsim/dart #548: Group/Metaskeleton semantics are not expressive enough.

mklingen, 2015-11-03:
Right now, a `Group` or `MetaSkeleton` is fundamentally a collection of DOFs. The `BodyNode`s in a group come along for free by virtue of being attached to the DOFs in the group. But what if I want other body nodes rigidly attached to the group by upstream or downstream fixed DOFs?

For example, suppose I want to get all the bodies in the "arm" of my robot. This should include the base of the arm, all of the joints of the arm, and then any attached bodies on the end effector. If I create a group with just the arm dofs, this excludes the base and the attached bodies.

Here's a diagram with the group selected by "*":

```
arm_base --> *joint_1* --> *link_1* --> ... --> *joint_N* --> *link_N* --> (downstream links/joints)
```

I would like to be able to add additional bodies to the group without adding their associated DOFs.

mkoval, 2015-11-04:
I've also run into this issue. The underlying problem is, like @mklingen says, is that a `MetaSkeleton` is really only a collection of `DegreeOfFreedom`s. As a result, it can be used to either represent: (1) a collection of `DegreeOfFreedom`s or (2) a collection of `BodyNode`s. It cannot simultaneously represent (1) and (2) at the same time because membership between `BodyNode`s and `DegreeOfFreedom`s are coupled.

In the past, I've worked around this by creating two separate `MetaSkeleton`s: one that I use as a collection of `BodyNode`s and one that I use as a collection of `DegreeOfFreedom`s. This gets the job done, but I agree with @mklingen that it's a bit awkward. It's also error-prone because the former collection contains an arbitrary set of `DegreeOfFreedom`s.

mxgrey, 2015-11-11:
I'm certainly open to changes in all of the MetaSkeleton related classes. If someone can put together a draft of the kind of API they would like, I could work off of that.

Part of the reasoning behind the current implementation is that it allows you to get dynamics information for subsets of Skeletons. It doesn't make much sense (to me, at least) to be able to decouple degrees of freedom from their bodies (or vice-versa) when talking about dynamics information. Another part of the reasoning is that it makes the dynamics computations fairly performant (the only overhead is copying data; it doesn't need to do any kind of sorting/collating).

But obviously there are applications for DART beyond dynamics, so it makes sense to offer greater freedom. I could imagine something like a ``BodyNodeGroup`` class, ``DegreeOfFreedomGroup`` class, and a class that concatenates the two while allowing the BodyNodes and DegreesOfFreedom to be managed independently. My questions in that case would be: What functions would you expect these classes to have? What would make the classes different from something like a ``std::vector<BodyNodePtr>`` or ``std::vector<DegreeOfFreedomPtr>``? Would you expect to still get the full ``MetaSkeleton`` API that offers all the various dynamics functions?

mklingen, 2015-11-11:
Here's what I want in a `Metaskeleton`:

* Everything I have now
* PLUS being able to add additional child bodies of the meta-skeleton (bodies *affected by* the joints already in the `MetaSkeleton`, but which are not directly connected to those joints).

I *don't* want to add arbitrary bodies or joints to the `MetaSkeleton` (such as ancestors of the `MetaSkeleton` or unrelated joints without their bodies). 

For example, I would like a metaskeleton called "Head" that contains the joints of my robot's head and all the children of those joints all the way to the leaves. I would like a metaskeleton called "right arm" that contains all the joints in the right arm plus all of the child bodies of those joints all the way to the leaves of the skeleton, etc.

mxgrey, 2015-11-11:
Do you want it to be strictly enforced that the additional bodies are descendants of at least one DOF that is currently in the group?

mklingen, 2015-11-11:
I don't think that will be necessary. Would adding additional bodies complicate any of the dynamics?

mxgrey, 2015-11-11:
When it comes to the ``ReferentialSkeleton``, the dynamics computations are really being done by the ``Skeleton`` that it's holding references to. The ``ReferentialSkeleton`` just repackages the information after it's been computed. The only complications that could arise would be with how the references are managed and how the information gets collated. But at the end of the day, these are just implementation details. I can already think of a way to handle this without very many drastic changes to the current implementation.

There are several different extensions of the ``ReferentialSkeleton`` class which each offer different restrictions on what sets of BodyNodes/DegreesOfFreedom are allowed inside of them. For example, the ``Branch`` class will start from a single BodyNode and is required to include all the BodyNodes and DegreesOfFreedom which descend from it. Would you like more flexibility in these classes that have these strict criteria? For example, would you like to be able to take a ``Branch`` and be able to remove certain ``DegreesOfFreedom``, so long as all the BodyNodes remain in the Branch? Or would you be happy enough to just have complete freedom with the ``Group`` class? (I should note that it is trivial to construct a ``Group`` that duplicates a ``Branch``)

mklingen, 2015-11-11:
I think that if `MetaSkeleton` or `Group` were just refactored to allow adding/removing either joints or bodies without adding/removing their respective bodies the rest would become easy to modify to suit my needs.

Right now, `getBodyNode(size_t i)` is implemented by getting `DegreeOfFreedom` at index `i` and grabbing the child body. If instead it were implemented as storing a separate vector of `DegreeOfFreedom` and `BodyNode` it would be possible to add/remove arbitrary DOFs or bodies. Then in the current `add/remove` functions for DOFs and bodies a parameter can be added to specify whether or not to remove the associated body/DOF. That would make the adding/removing a bit slower but would allow more flexibility.

mxgrey, 2015-11-16:
I wanted to get your thoughts on a few more details as I work on this feature request:

In the current DART semantics, calling ``->getBodyNode(i)->getParentJoint()`` is always equivalent to calling ``->getJoint(i)``. This is true for all the currently existing ``MetaSkeleton`` implementations. Would it be okay to keep this behavior, or would you want to be able to include a ``BodyNode`` in a ``Group`` without including its parent ``Joint`` in that ``Group``?

I suspect the most sensible thing would be to give the user complete control over which BodyNodes, Joints, and DegreesOfFreedom are included in the Group, and there should not be any strong coupling between these things. This would require a few more significant changes to the ReferentialSkeleton, but it might be worth it. I'm not sure, though, since I personally always use whole Skeletons and therefore don't have strong opinions on what exactly ``ReferentialSkeleton`` should offer.

mklingen, 2015-11-17:
I agree. A group should just be an arbitrary collection of DOF and BodyNode.

mxgrey, 2015-11-18:
I'm still not clear on how you would prefer ``Joint`` to be handled.

Just to clarify: Every ``BodyNode`` has exactly one parent ``Joint``, and each ``Joint`` has some fixed number of ``DegreeOfFreedom``s based on what type of Joint it is. The ``MetaSkeleton`` API offers the functions ``getBodyNode(size_t)``, ``getJoint(size_t)``, and ``getDof(size_t)``. It's clear now that for the ``Group`` implementation of the ``MetaSkeleton``, there should be no coupling between the ``BodyNode``s and the ``DegreeOfFreedom``s that are present in a Group, so ``Group::getBodyNode(size_t)`` will be completely independent of ``Group::getDof(size_t)``. But currently, calling ``Group::getJoint(size_t)`` is exactly the same as calling ``Group::getBodyNode(size_t)->getParentJoint()``, meaning that the ``BodyNode``s in a ``Group`` are strongly coupled with the ``Joint``s in that ``Group``.

So the question I would have is whether we should change this so that the user can decide which ``Joint``s are in a ``Group`` independently of which ``BodyNode``s are in a ``Group``, such that ``Group::getJoint(size_t)`` will not necessarily be equivalent to ``Group::getBodyNode(size_t)->getParentJoint()``.

mklingen, 2015-11-18:
Interesting. I forgot that joints may have more than one DOF. I think its
pretty clear that `getJoint(i)` should return the `ith` joint, not the
joint associated with the `ith` body, if you're going to be adding Joints
arbitrarily.

mxgrey, 2015-11-18:
> if you're going to be adding Joints arbitrarily.

Right, but currently you cannot add a ``Joint`` arbitrarily; it gets added along with its child ``BodyNode``. So I get the impression you think that it should be possible to add and remove ``Joint``s arbitrarily, independently of their child ``BodyNode``s?

If so, then this raises one more question: When a ``Joint`` gets added to a ``Group`` should all of the ``DegreeOfFreedom``s that belong to it be added as well? Or should we have a flag for that, such as ``Group::addJoint(Joint* joint, bool addDofs = true)``?

mklingen, 2015-11-18:
Yes, and I think it should add all of the DOFs as well. Either that or we
don't allow adding joints, just DOFs and bodies.

mxgrey, 2015-11-18:
I think I'll have the ``addJoint`` function contain a flag for adding all of its DOFs, and the flag will default to being true (as seen in the example of my last post), unless you have a reason to strongly object to that.

mkoval, 2015-11-18:
> In the current DART semantics, calling `->getJoint(i)` is always equivalent to calling `->getBodyNode(i)->getParentJoint()`. This is true for all the currently existing `MetaSkeleton` implementations. Would it be okay to keep this behavior, or would you want to be able to include a `BodyNode` in a `Group` without including its parent `Joint` in that `Group`?

I didn't even realize this was the case, so I definitely do not rely on this behavior. The explicit version `->getBodyNode(i)->getParentJoint()` seems more straightforward than `getJoint(i)`, at least to me, even if they are equivalent.

>  I'm not sure, though, since I personally always use whole `Skeleton`s and therefore don't have strong opinions on what exactly `ReferentialSkeleton` should offer.

In motion planning there are generally: (1) a set of `DegreeOfFreedom`s to plan for and (2) a set of `BodyNode`s to collision check. Often times (2) consists of the set of all `BodyNode`s that can move by changing the position any `DegreeOfFreedom` in (1). Since a `MetaSkeleton` contains both `BodyNode`s and `DegreeOfFreedom`s, I think it would be useful to make the two sets completely independent.

> If so, then this raises one more question: When a `Joint` gets added to a `Group` should all of the `DegreeOfFreedom`s that belong to it be added as well? Or should we have a flag for that, such as `Group::addJoint(Joint* joint, bool addDofs = true)`?

I don't have a strong opinion about this either way. From a motion planning perspective, `Joint`s don't really matter - only their `DegreeOfFreedom`s do. I see four possibilities:

1. completely decouple the `Joint`s from the `BodyNode`s and `DegreeOfFreedom`s
2. include any `Joint`s that owns a `DegreeOfFreedom` in the `MetaSkeleton`
3. include the parent joint of any `BodyNode` in the `MetaSkeleton`
4. remove the `getJoints` function entirely

I don't feel strongly about this, but have a slight preference for (2). I think (3) is superfluous because you can easily construct the same set of `Joint`s by iterating over the `MetaSkeleton`'s `BodyNode`s and calling `getParentJoint`. At the end of the day, it depends what you think people will use `MetaSkeleton::getJoints` for.

mxgrey, 2015-11-18:
> At the end of the day, it depends what you think people will use ``MetaSkeleton::getJoints`` for.

I agree that this is the central question. The truth is, I don't really know. I don't actually use the ``MetaSkeleton`` myself, since I always work in terms of entire Skeletons.

My main concern about option (2) is that it implies that a Joint should not be included if it is zero-dof. The ``Joint`` class *does* have a few properties that are not directly accessible through its DOFs, most notably the transform-from-parent and the transform-from-child properties, so you might want to have ``Joint``s directly accessible through a ``Group``/``MetaSkeleton`` if you intend to do operations on those properties.

I think my strongest preference right now would be to allow the user **complete** control over what gets included in a ``Group``, but design the API with reasonable default behavior (i.e. adding a Joint will also add all of its DOFs by default, and adding a DOF will also add its Joint by default).

mkoval, 2015-11-18:
> My main concern about option (2) is that it implies that a Joint should not be included if it is zero-dof.

I have the same concern about (2).

> I think my strongest preference right now would be to allow the user complete control over what gets included in a `Group`, but design the API with reasonable default behavior (i.e. adding a `Joint` will also add all of its DOFs by default, and adding a DOF will also add its `Joint` by default).

That's fine by me. I think it's the best of mediocre options.

mxgrey, 2015-11-18:
Would it make more sense to change the API to be like

(1) ``Group::addDof(DegreeOfFreedom* dof, bool addJoint=true, bool warning=true)`` 
or (2) ``Group::addDof(DegreeOfFreedom* dof, bool warning=true, bool addJoint=true)``

I think option (1) would make more sense to me, but option (2) would be a perfectly backwards compatible API change. If anyone has opinions on this, I'd like to hear them.

mxgrey, 2015-11-24:
This feature request should be addressed as of #557. Let me know if the changes there are not sufficient.

closed 2015-11-24 by mxgrey