dartsim/dart #549: Undefined reference to Assimp symbol

mkoval, 2015-11-04:
When building DART (and another project that depends on it) with GCC, I get this linking error referring to the `Assimp::IOSystem::ComparePaths(char const*, char const*) const` function:
```
undefined reference to symbol '_ZNK6Assimp8IOSystem12ComparePathsEPKcS2_'
```

I can fix this by manually linking my project against `assimp`. The obvious solution is to make the  `DART_LIBRARIES` variable exported by `DARTConfig.cmake` include `ASSIMP_LIBRARIES` (similar to how `DART_INCLUDE_DIRS` includes `EIGEN3_INCLUDE_DIRS`).

However, I don't think this should be necessary. I suspect that this is caused by the same root issue as #451 that we discussed in #452, #455, #459, and #542.

jslee02, 2018-02-13:
Need to check if this is persistent.

jslee02, 2019-04-29:
Closing since I cannot reproduce this error. Feel free to reopen if this is still persistent.

closed 2019-04-29 by jslee02