dartsim/dart #567: Cannot compile DART on Windows 7 / Visual Studio 2013 for x64

se210, 2015-12-07:
I'm having trouble compiling DART on Windows 7 64-bit and Visual Studio 2013 32-bit for x64. I followed the instructions on the Wiki for Windows compilation (https://github.com/dartsim/dart/wiki/DART%205.1%20Installation%20for%20Windows), but I keep getting crashes in cl.exe with error message "Microsft (R) C/C++ Optimizing Compiler has stopped working." Attached is an build output from Visual Studio. I did not install nlopt.

[output.txt](https://github.com/dartsim/dart/files/54714/output.txt)

jslee02, 2015-12-07:
According to `output.txt`, it seems you have old version of DART installed in `C:\Golems\include\dart`. 
```
7>          C:\Golems\include\dart/utils/Parser.h(95) : see declaration of 'dart::utils::openXMLFile'
```

Please rebuild DART after remove it.

se210, 2015-12-08:
I think the older version of DART headers were installed by downloading and installing [DART 5.0-prerequisites-msvc12-md-32bit.msi] (https://github.com/dartsim/dart-prerequisites-windows/raw/3e34167fd29fcc870ebed50bf318f3f31f491e6a/DART%205.0-prerequisites-msvc12-md-32bit.msi) from the installation instruction. I tried manually deleting ```C:\Golems\include\dart```, but it still seems to be having the same issue.

[output2.txt](https://github.com/dartsim/dart/files/54906/output2.txt)

Message from first instance of the compiler crashing says:
5>C:\Users\Administrator\Downloads\dart-master\dart\utils\VskParser.cpp(380): fatal error C1001: An internal error has occurred in the compiler.
5>  (compiler file 'msc1.cpp', line 1325)
5>   To work around this problem, try simplifying or changing the program near the locations listed above.
5>  Please choose the Technical Support command on the Visual C++ 
5>   Help menu, or open the Technical Support help file for more information
5>cl : Command line error D8040: error creating or communicating with child process

se210, 2015-12-09:
Were you able to reproduce the issue? I tried on another machine and it has the same symptoms. I tried both Win32 and Win64 builds with Visual Studio 2013 on Windows 7 64bit.

jslee02, 2015-12-09:
I tested on my Windows 10 64-bit machine with Visual Studio 32-bit (building 32-bit binaries), but couldn't find any problem.

`DART 5.0-prerequisites-msvc12-md-32bit.msi` doesn't install DART headers into `C:\Golems\`. Could you check it again by uninstalling `DART 5.0-prerequisites-msvc12-md-32bit.msi`, removing `C:\Golems\`, and reinstalling `DART 5.0-prerequisites-msvc12-md-32bit.msi` without building DART yet?

One possible way of that old version of DART headers are copied to `C:\Golems\` is you run INSTALL project in dart solution of an old version of DART. Whether or not, the headers won't be installed by `DART 5.0-prerequisites-msvc12-md-32bit.msi`.

I would like to check if you are trying to build 32-bit binaries or 64-bit binaries. You might not be able to build 64-bit binaries with the DART dependencies built as 32-bit. To build 64-bit binaries, you need 64-bit dependencies.

Also, this might not the case, but please make sure your Visual Studio 2013 is up to date.

By the way, I don't have a machine Windows 7 64-bit installed. I would appreciate that if any one on Windows 7 64-bit couldn test this.

se210, 2015-12-10:
Ah, thank you for your suggestion to check Visual Studio 2013 version! It turns out VS2013 doesn't update itself and I had to manually update it. After updating to Visual Studio 2013 Update 5, 32-bit binary compiles fine.

However, I would like to still compile a 64-bit binary. Do you have a 64-bit version of dependencies available?

jslee02, 2015-12-10:
Glad to hear it worked out!

Currently, we don't have 64-bit versions yet. It's not easy to manage all the versions like for 32/64-bit, MD/MT, and Visual Studio 2013/2015. So I'm looking forward an available package manager for Windows (or even cross platform) that takes the source code and provides a various version of binaries like `apt` and `homebrew`. I'm [testing](https://github.com/jslee02/conan-dart) [`Conan.io`](https://conan.io/), but not sure when it's ready. I also tried `chocolatey` and `nuget`, but they doesn't seem to fit our purpose; they require compiled ones instead of the source code.

In the meantime, you might need to compile the 64-bit binary for your use. Sorry for the convenience.

closed 2015-12-10 by se210