dartsim/dart #568: Proposal: Use .hpp for headers when switching to the new package name

mxgrey, 2015-12-08:
In various discussions, I've heard interest in the idea of having C-bindings for DART since this would enable easier interfacing with other programming languages, and therefore easier integration into other frameworks. If someone were to implement C-bindings, the most natural file organization would be to have C API headers use ``.h`` while C++ API headers use ``.hpp``, but in our current file organization all the C++ headers are already using ``.h``.

Whether or not anyone is currently planning to implement C-bindings, I think it would be good to use the ``.hpp`` convention to allow for a clear delineation between the existing C++ API and a C API in case one is ever created. Since the package will be undergoing a highly interruptive name change in the near future anyway, I think that would be the best possible time to make a change like this.

jslee02, 2015-12-08:
:+1: It sounds reasonable.

jslee02, 2016-04-26:
Addressed by #693

closed 2016-04-26 by jslee02