dartsim/dart #575: Some typo error in tutorial dominoes

JiaYeLi, 2015-12-23:
http://dart.readthedocs.org/en/release-5.1/tutorials/dominoes/

in the Overview
Write a stable PD controller w/ gravity and coriolis compensation

I guess "w/" is with but I'm not sure

mxgrey, 2015-12-23:
``w/`` is a common abbreviation for the word ``with``. It's not actually a typo, but it may be worth changing it to ``with`` explicitly if not everyone is familiar with that abbreviation.

JiaYeLi, 2015-12-24:
Thank you, mxgrey.

I learned a new abbreviation.
I am sorry that I'm not a native speaker of English.

mxgrey, 2015-12-24:
We'll definitely want DART and its documentation to be accessible to non-native English speakers, so I'm actually glad that you're not a native speaker of English; that will help us identify spots in the documentation that might be confusing or unclear for people who are not native speakers.

jslee02, 2016-04-16:
I believe we can close this issue. Please re-open if not the case.

closed 2016-04-16 by jslee02