dartsim/dart #583: Possible memory leaks in constraint solver?

costashatz, 2016-01-12:
Hello,

After running huge simulations in our cluster, we observed some memory leaks that filled up our RAM and swap (64gb + 64gb!). I was able to create a small program that `valgrind` finds possible memory leaks:

```
==7983== Memcheck, a memory error detector
==7983== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==7983== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==7983== Command: ./hexapod
==7983== 
==7983== 
==7983== HEAP SUMMARY:
==7983==     in use at exit: 4,851 bytes in 42 blocks
==7983==   total heap usage: 157,188 allocs, 157,146 frees, 178,053,923 bytes allocated
==7983== 
==7983== 4,720 (1,760 direct, 2,960 indirect) bytes in 10 blocks are definitely lost in loss record 6 of 6
==7983==    at 0x4C2BBA0: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==7983==    by 0x54206E1: dart::constraint::ConstraintSolver::updateConstraints() (in /usr/local/lib/libdart-core.so.5.1.1)
==7983==    by 0x5420D5F: dart::constraint::ConstraintSolver::solve() (in /usr/local/lib/libdart-core.so.5.1.1)
==7983==    by 0x5426522: dart::simulation::World::step(bool) (in /usr/local/lib/libdart-core.so.5.1.1)
==7983==    by 0x43AB1A: main (in /home/kchatzil/ownCloud/documents/backup-2015-12-10/Workspaces/DART/source/dart_test/build/hexapod)
==7983== 
==7983== LEAK SUMMARY:
==7983==    definitely lost: 1,760 bytes in 10 blocks
==7983==    indirectly lost: 2,960 bytes in 30 blocks
==7983==      possibly lost: 0 bytes in 0 blocks
==7983==    still reachable: 131 bytes in 2 blocks
==7983==         suppressed: 0 bytes in 0 blocks
==7983== Reachable blocks (those to which a pointer was found) are not shown.
==7983== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==7983== 
==7983== For counts of detected and suppressed errors, rerun with: -v
==7983== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

Just running a few simulations does not make a problem, but we are using DART in evolutionary algorithms and we are running million of small simulations. So, small memory leaks add up and create a problem.

I am using DARTCollisionDetector, but I also tried with FCL and Bullet and I still get similar memory leak errors.

Any ideas on how to solve this issue?

*Sorry for the long question*. The small simulation code I am using is the following (basically code from your tutorials):

```cpp
#include "dart/dart.h"
#include "dart/collision/dart/DARTCollisionDetector.h"
#include "controllerDuty.hpp"

using namespace dart::common;
using namespace dart::dynamics;
using namespace dart::simulation;
using namespace dart::math;

class Controller {
public:
    Controller(const SkeletonPtr& pexod)
        : mPexod(pexod)
    {
        std::vector<float> reference_ = std::vector<float>(36);
        reference_ = {1, 0, 0.5, 0.25, 0.25, 0.5, 1, 0.5, 0.5, 0.25, 0.75, 0.5, 1, 0, 0.5, 0.25, 0.25, 0.5, 1, 0, 0.5, 0.25, 0.75, 0.5, 1, 0.5, 0.5, 0.25, 0.25, 0.5, 1, 0, 0.5, 0.25, 0.75, 0.5};
        controller_ = new ControllerDuty(reference_, std::vector<int>());
        // Grab the current joint angles to use them as desired angles
        mQDesired = mPexod->getPositions();

        size_t nDofs = mPexod->getNumDofs();

        mForces = Eigen::VectorXd::Zero(nDofs);

        // // Set PD control gains
        // mKpPD = 200.0;
[... 1088 lines cut ...]
      <origin rpy="1.57079632679 0 0" xyz="0.0125 -0.0475 0"/>
      <geometry>
        <cylinder length="0.095" radius="0.025"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="1.57079632679 0 0" xyz="0.0125 -0.0725 0"/>
      <geometry>
        <sphere radius="0.025"/>
      </geometry>
      <material name="Red"/>
    </collision>
    <inertial>
      <!-- CENTER OF MASS -->
      <origin rpy="1.57079632679 0 0" xyz="0.0125 -0.0475 0"/>
      <mass value="0.184"/>
      <!-- Cylinder inertia: 1/12*m*(3*r^2+h^2), 1/12*m*(3*r^2+h^2), 1/12*m*r^2 -->
      <inertia ixx="0.000129183333333" ixy="0" ixz="0" iyy="0.000129183333333" iyz="0" izz="6.13333333333e-06"/>
    </inertial>
  </link>
  <!-- END OF LEG LINKS/JOINTS -->
  <!-- <xacro:include filename="$(find pexod_description)/urdf/modules/imu.xacro" /> -->
  <!-- <xacro:include filename="$(find pexod_description)/urdf/modules/cameras.xacro" /> -->
</robot>
```

mxgrey, 2016-01-12:
Thanks for reporting this issue. I recommend using gist.github.com when you want to include long snippets of text.

Unfortunately, not quite **all** of DART's codebase has been updated to using C++11 yet, so that leaves us open to some memory leak vulnerabilities. I'm noticing that several vectors of dynamically allocated constraints are not being properly deleted in the destructor of the ``ConstraintSolver`` class, and that could certainly explain the leak you're seeing. I'll try switching to shared_ptrs for these constraints, and that should remedy the memory leak.

Just to clarify, are you frequently creating and destroying ``World`` objects (and the Skeletons inside of them), or do you recycle ``World`` objects so that you don't need to continuously create and destroy them? If you are recycling your ``World`` objects, then there must be another cause behind the memory leaks that I haven't identified yet. Switching to ``shared_ptr`` should still fix it, but I'd be curious to know where else it might have been leaking.

Also, if you are frequently creating and destroying ``Skeleton``s with meshes, then that might result in [a known memory leak caused by Assimp](https://github.com/dartsim/dart/issues/565). If you are not using any meshes, then you shouldn't need to worry about that one.

costashatz, 2016-01-12:
> Unfortunately, not quite all of DART's codebase has been updated to using C++11 yet, so that leaves us open to some memory leak vulnerabilities. I'm noticing that several vectors of dynamically [...]

I am waiting for this. When can you have this ready? I can help you on that one, if you can point to me where I should look.

> Just to clarify, are you frequently creating and destroying World objects (and the Skeletons inside of them), or do you recycle World objects so that you don't need to continuously create and [...]

I am not recycling World objects. I am creating and destroying them every time.

> Also, if you are frequently creating and destroying Skeletons with meshes, then that might result in a known memory leak caused by Assimp. If you are not using any meshes, then you shouldn't need to worry about that one.

I am loading a skeleton once (using URDF file) and then cloning it each time. Inside URDF I only have basic shapes (spheres, cylinders, boxes). So I guess I shouldn't worry about that.

mxgrey, 2016-01-12:
> I am waiting for this. When can you have this ready?

I'm making the changes right now. It should be done some time this afternoon.

mxgrey, 2016-01-12:
Although to clarify, for right now I'm only going to focus on changing the known trouble spots in ``ConstraintSolver``. Attempting to scan through the whole code base for areas that are not yet updated to modern C++ practices would take much longer.

costashatz, 2016-01-12:
> I'm making the changes right now. It should be done some time this afternoon.

Thanks for prompt answer once again.

mxgrey, 2016-01-12:
I've created a pull request #584 with the changes. I have not yet run any tests to make sure that the leaking is gone, but I'd be very surprised if it still persists (unless there's another leak source that wasn't caught in the Valgrind report).

costashatz, 2016-01-14:
This was solved by #584 !

closed 2016-01-14 by costashatz