dartsim/dart #594: Add Texture to shape

costashatz, 2016-01-22:
Hello,

I was wondering if it's possible to add texturing to a dart shape (e.g. BoxShape).

mxgrey, 2016-01-22:
Textures are definitely possible with ``MeshShape``s. Are you using the native DART renderer, or are you using a wrapper, like Gazebo? I don't know offhand whether Gazebo supports the textures in DART's ``MeshShape`` class.

If you're using DART's native rendering, then I *think* the following approach should work (although I haven't tested it):

1. Create a textured box mesh in a 3D modelling program such as Blender
2. Load and add the mesh as a **visualization** shape for the BodyNode
3. Construct a BoxShape primitive and add it as a **collision** shape for the BodyNode

That should let you get the visual results that you want while still having good performance when it comes to collision detection.

costashatz, 2018-02-13:
I think this is solved with #973 ...

closed 2018-02-13 by costashatz