dartsim/dart #597: In debug mode, we should throw assertions when incoming positions are NaN or Inf

mxgrey, 2016-01-28:
Currently we have assertions which (in debug mode) check that the transformations generated by the kinematics and dynamics are valid transformations, and this is useful for identifying when a bug is occurring. But by the time we're computing the transformation, it can be hard or even impossible to tell what the source of the bug was.

I think if we put in assertions to check that incoming positions (and velocities, accelerations, etc) are not NaN or Inf, it will be much easier to identify what the source of the error is.

jslee02, 2016-01-28:
Precondition check sounds good. I'd like to keep the current postcondition checks along precondition checks.

mxgrey, 2016-01-28:
Yes, I agree that we should keep all the existing checks in place.

closed 2025-11-26 by jslee02