dartsim/dart #609: Markers should be Nodes

mxgrey, 2016-02-13:
I think we should make Markers into Nodes so that they can be managed by the Node framework. I believe this has been discussed before, but I wanted to raise it as an issue to help us remember to do it.

mkoval, 2016-02-13:
I thought that `Marker` was superseded by `ShapeNode` for visualization. What is the difference between a `Marker` and a `ShapeNode` with only a `VisualAddon` attached?

jslee02, 2016-02-13:
I think the differences are (1) it's a point rather than a frame and (2) it has `ConstraintType`. Since we don't a class for a pointer, it would be just `ShapeNode` + `VisualAddon` + `ConstraintType`.

We can consider having `Point` class that is similar to `Frame`, but it would only have a relative position from its parent `Frame` instead of a relative transformation. Also, I think `Point` class should be a leaf in the kinematic tree since it doesn't provide the reference frame for its child `Frame` or `Point`.

Anyways, :+1: to the idea of making `Marker` as `Node`.

mkoval, 2016-02-13:
Can you explain what a `Marker` is? Also, what is a `ConstraintType`?

jslee02, 2016-04-27:
> Can you explain what a Marker is? Also, what is a ConstraintType?

Sorry for late answer. I believe `Marker` has been used to represent markers that is parsed from motion formats like VSK. `ConstraintType` is a utility data structure for the case that the marker is used in IK like `EndEffector` to represent the restriction level of the constraint.

---

Closing this as resolved by #692

closed 2016-04-27 by jslee02