dartsim/dart #610: DART 5.1 can't be built with VS2015 on 32-bit Windows

jslee02, 2016-02-13:
As reported in #606, DART 5.1 can't be built with VS2015 o n32-bit Windows. It seems Visual Studio can't use enough heap memory for parsing headers so optimizing DART header organization for simpler header dependencies would be a possible solution.

JiaYeLi, 2016-05-02:
I set "Maximum number of parallel project builds" to 1 and I build Win32 project successfully with VS2015(32-bit) on 64-bit Windows (Windows 10 and Windows 8.1).

I think it may be an alternative way to get the Win32 DART 5.1

----
How to set Maximum number of parallel project builds 
https://msdn.microsoft.com/en-us/library/y0xettzf(v=vs.90).aspx
----
By the way.
If I set "Maximum number of parallel project builds" to 8, my computer will exhausts 8GB memory.

I am not sure the reason.
I think there are too many "Microsoft C/C++ Compiler Driver" processes running simultaneously.

jslee02, 2018-02-13:
Closing as we don't support 5.x anymore.

closed 2018-02-13 by jslee02