dartsim/dart #611: Could Actuators be associated with DOF rather than Joint?

tianyang9310, 2016-02-14:
Now DART can only set actuator type for each joint, is it possible that we can set different actuator type for different DOF within the same joint? # @

mxgrey, 2016-02-15:
@jslee02 can correct me if I'm wrong, but I believe that the hybrid articulated dynamics algorithm can only operate on a per-joint basis because of the coupling between degrees of freedom for certain joints. For example, the degrees of freedom in BallJoint and the first three in FreeJoint are deeply intertwined.

You could still accomplish what you're looking for by breaking apart a multi-dof joint into multiple single-dof joints. For example, a FreeJoint could be made into three prismatic joints followed by three revolute joints. In between each joint, you should place a BodyNode without any collision geometry or mass. Ordinarily, I would say to be extremely wary of having BodyNodes without any mass, but as long as they're in a chain between two masses, it should be okay.

jslee02, 2016-02-15:
That's correct. One possible convenient API would be a composite joint class, which a joint consists of multiple serial joints with zero mass and inertia bodies between the joints as @mxgrey said. But @mxgrey 's suggestion would be the quickest solution for now since it's unsure when or whether the composite joint class is implemented not yet.

psigen, 2016-02-15:
Hmm, is there some reason to create a composite joint?  This seems like a fairly heavyweight structure to maintain instead of simply making sequential joints.

Is there some efficiency reason why sequential joints are not desirable?

mxgrey, 2016-02-15:
I wouldn't recommend having a composite joint class. I feel like that would introduce unnecessary complexity. Perhaps having a convenience function that generates a sequence of joints with massless bodies in between would be okay, but I'd personally object to that, because I don't want to encourage users to have Skeletons with massless bodies unless they *really* know what they're doing and why they're doing it.

psigen, 2016-02-15:
@mxgrey: This might be a bit off-topic, but what's the preferred mechanism for having intermediate attachments points that are massless?

Suppose I have a large aluminum block (e.g. Herb's head) that has a number of fixed attachment points on it (like a few camera mounts), and I have some cameras with several attachment points (e.g. one on each side).  Presumably when I attach these two things together, I shouldn't lose the intermediate attachment information, and should be able to compute things in those attachment frames, right?

jslee02, 2016-02-15:
The composite joint idea would be two benefits: (1) It would provide a convenient way to create sequential joints automatically inside the composite joint instead of letting the user create the sequential joints (and dummy bodies) manually. At the same time, the user can handle the composite joint as a single joint. This will prevent to have different joint instances for the actual single joint. (2) It would enable us to create more kinds of joint types by combinating current joint types.

Yeah, it would bring much complexity. Utility function that generates the serial joints would be enough unless we really want the functionality of different joint actuator types per dof in one joint.

psigen, 2016-02-15:
@jslee02 It seems like you would have to provide the same amount of information (i.e. relative joint axis transformations for each sequential joint) to create the composite joint.  But if this is the case, then it seems more useful to have a sequence of joints, so that the intermediate joint origin data can be queried as part of the kinematic tree.

I agree that convenience utilities for creating common sequences of joints would potentially be useful.

mxgrey, 2016-02-15:
@jslee02 I guess if the composite joint is seen as a single (multi-dof) joint from the point of view of the Skeleton, I might be okay with that. That would guarantee that the user doesn't accidentally split the sequence of joints in half, leaving massless bits dangling off the end of the tree.

@psigen Having massless attachment points as leaf nodes would be problematic if put into a simulation, because it would result in infinite accelerations. One viable option might be to have a Node class that can be used as a factory for generating joint-body attachments. For example, we could have an ``AttachmentNode`` class that contains joint properties ready to go, and provides an API for attaching BodyNodes based on those properties with functions like ``createJointAndBodyNodePair`` and ``attach``.

psigen, 2016-02-15:
@mxgrey: I think API-wise it might be best to have a massless Node that can be put in between `Joint`s and `BodyNode`s, but is not used for simulation (basically just storing a rigid transformation to the next BodyNode).  Let's discuss further offline.

jslee02, 2016-02-15:
Right, it would require the same amount of information. The composite joint would be like a wrapper joint  that inherit `Joint` so it is seen as a multidof single joint as @mxgrey said. It would actually contain the serial joint instance in it and would be able to extract the original data as well.

Anyways, it seems we all agree to have utility functions instead of composite joint class.

mxgrey, 2016-02-15:
I would actually be okay with the CompositeJoint class that's been described. But I'd consider it to be pretty low priority at the moment.

jslee02, 2016-02-15:
I agree. We can continue to discuss to seek better solution once other high priority changes are done (at least after the next major release).

mkoval, 2016-02-15:
I also *strongly* prefer a chain of single-DOF `Joint`s connected by massless `BodyNode`s. I see very little to introducing `CompositeJoint`. To address @jslee02's specific points:

> It would provide a convenient way to create sequential joints automatically inside the composite joint
> instead of letting the user create the sequential joints (and dummy bodies) manually.

Like @psigen pointed out, the user still has to provide `Properties` for each `Joint`. At this point, they might as well call the constructor, too.

> It would enable us to create more kinds of joint types by combining current joint types.

I see this as a downside. It adds complexity without adding any new capabilities.

---

I think the key, reoccurring issue, is that *massless `Node`s are a fact of life*. I understand that this is unnerving from a simulation perspective, for the reasons that @mxgrey outlined, but disallowing massless `BodyNodes`:

- is the primary reason why an arbitrary `Node` may not have `BodyNode` children
- prevents the user from defining intermediate frames like those described by @psigen 
- complicates loading, since most URDF files contain massless `<link>`s for this very purpose.

I think that we should find a way of gracefully handling massless `Node`s in the dynamics engine, rather than further contorting the `BodyNode` API to pretend that they do not exist.

---

My understanding is the constraint we really want to enforce is that:

> 1. Every non-trivial `Joint`, i.e. non-`WeldJoint`, has a subtree that contains non-zero mass.
> 2. Every `BodyNode` with collision geometry has a subtree that contains non-zero mass.

This is quite a bit looser than the constraints that DART currently enforces. Hopefully @mxgrey and @jslee02 can verify that I'm not missing any cases here.

tianyang9310, 2016-02-16:
@jslee02 For now I have figured out another way to fulfill my goal. Basically I want to set acceleration and velocity command to the two DOF of a planar joint. If I set them individually then the former setting would be overridden by the latter one. Except the sequential joints with massless BodyNode solution you mentioned above, I do it by integrating acceleration into velocity at each time step (using timestep of the world class 'world->getTimestep()') and only setting the joint actuator type as VELOCITY. Please correct me if I should use other timestep since there are several time clock systems in DART. 

Another issue is that the SERVO actuator type seems not working at all. From what I am supposing, SERVO should work like VELOCITY except that it meets the joint constraints. But in my simulation when I set SERVO actuator type, it doesn't work at all. My DART version is release 5.1 and my OS is MAC OS.

But anyway, thank everyone above for helping me.

jslee02, 2016-02-17:
> For now I have figured out another way to fulfill my goal. Basically I want to set acceleration and velocity command to the two DOF of a planar joint. If I set them individually then the former setting would be overridden by the latter one.

If your joint actuator type is VELOCITY, then DART overrides the acceleration as `acc = (desiried_vel - previous_vel) / dt`, which would make sense kinematically as the current acceleration should be consistent with the previous velocity and desired velocity. If you just want to set arbitrary joint acceleration (and velocity that is integrated with the acceleration), then you might want to use ACCELERATION actuator type. It would also override the velocity, but the overridden velocity is just the integrated one with your acceleration. That means you don't need to set velocity but just acceleration in this case.

> Another issue is that the SERVO actuator type seems not working at all.

Servo actuator is not supported for DART 5.1, but it will be in DART 6.0 [Here](https://github.com/dartsim/dart/pull/566) is the PR for it. It seems this can be backported to DART 5.x, but I'm not sure when I'm available to do this.

> Please correct me if I should use other timestep since there are several time clock systems in DART.

DART allows skeletons to have individual timestep, but if the skeleton is added to a world then the timestep of the skeleton is set to the timestep of the world. So they would have the same timestep.

closed 2016-02-26 by tianyang9310